kriswiner/EM7180_SENtral_sensor_hub #5: EM7180+BMP280, Rotation along one axis results in significant rotation along other 2 axes as well.

mohit5037, 2016-03-08:
Hi Kris,

I bought this EM7180 a few days back. I read in its datasheet that you don't need to calibrate magneto in it and I have uploaded your code which has pre-written function for calibration of accelero and gyro but still rotation along any of the axes causes significant rotation along other axes.

I don't know what to send you as a proof but if you haven't seen this before I can send you the video of this process.

I am taking quaternions from hardware (Sentral) and then rendering a 3-D object in matlab. I am displaying quaternions as well as angles in matlab. 

Pls see the below video.
https://youtu.be/1iKlh5R45As

In this video I am making pitch to go from 0 to 90 but as you can see yaw and roll along changes very significantly.

Here are the logs:
Scanning...
I2C device found at address 0x28  !
done

EM7180 ROM Version: 0xE69
Should be: 0xE609
EM7180 RAM Version: 0x17435
EM7180 ProductID: 0x80 Should be: 0x80
EM7180 RevisionID: 0x2 Should be: 0x02
A barometer is installed
A temperature sensor is installed
EEPROM detected on the sensor bus!
EEPROM uploaded config file!
EM7180 in initialized state!
EEPROM upload successful!
Beginning Parameter Adjustments
Magnetometer Default Full Scale Range: +/-1000uT
Accelerometer Default Full Scale Range: +/-4g
Gyroscope Default Full Scale Range: +/-2000dps
Magnetometer New Full Scale Range: +/-1000uT
Accelerometer New Full Scale Range: +/-8g
Gyroscope New Full Scale Range: +/-2000dps
 EM7180 run status = normal mode
 EM7180 new quaternion result
 EM7180 new mag result
 EM7180 new accel result
 EM7180 new gyro result
 EM7180 sensor status = 0
Actual MagRate = 100 Hz
Actual AccelRate = 200 Hz
Actual GyroRate = 200 Hz
Actual BaroRate = 50 Hz
quat	0.00	0.00	0.57	0.82
quat	0.00	0.00	0.57	0.82
quat	0.00	0.00	0.57	0.82
quat	0.00	0.00	0.57	0.82
quat	0.00	0.00	0.57	0.82
quat	0.00	0.00	0.57	0.82
quat	0.00	0.00	0.57	0.82

Please correct me if I am doing something wrong here.

kriswiner, 2016-03-08:
The device has an auto calibration function but you have to wave it around in a figure eight pattern for it to work. Once it is calibrated you can save the calibration as warm start parameters on the on-board EEPROM. This sketch has the coding to do that written for the EM7180+MPU9250; you will have to modify your EM7180+BMX055 sketch accordingly.

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/tree/master/WarmStart

mohit5037, 2016-03-09:
Hi Kris,

Thanks for the early response. I have EM7810 + MPU9250 + BMP280 sensor with me.

I tried running the code you sent above. Given the following params:

passThrough = false;
warmStart = 0;

It does not stop for the magnetometer calibration for me to wave pattern eight. If I make passthrough = True then it stops for calibration but then I am not able to save those params in EEProm by entering 1 because you are reading that input inside of the first "if" block in loop function which belongs to passthrough = false;

kriswiner, 2016-03-09:
> axis results in significant rotation along other 2 axes as well. (#5)

Use the passthru = false (run in master mode, and move the magnetometer in a
figure eight motion until the device is calibrated. The mag calibration is
automatic; it doesn't wait for you to do anything it is calibrating all the
time. Butif you wave it around to sample the response surface it can adjust
the biases, etc. Once the mag reading is correct for your area, or the
Madgwick and SENtral Euler angles are about the same and the heading changes
by 90 degrees when the sensor is turned by ninety degrees you can press 1 to
save the parameters in the warm start parameters in the EEPROM.

<https://github.com/kriswiner/EM7180_SENtral_sensor_hub/issues/5#issuecommen
t-194128214> .
<https://github.com/notifications/beacon/AGY1qh3X2MQVI5df2hw1ke7Y2zoNaDwtks5
prmKLgaJpZM4HruZY.gif>

mohit5037, 2016-03-09:
I had tried this earlier. Is there any particular method to wave the sensor in eight pattern which I should follow?

kriswiner, 2016-03-09:
You just want to have the sensor sample as much of the response surface as
you can. Shouldn't take more than a minute or so.

<https://github.com/kriswiner/EM7180_SENtral_sensor_hub/issues/5#issuecommen
t-194151413> .
<https://github.com/notifications/beacon/AGY1qld4ZK4Bk5pA7I5bZM5XmcvWd7pKks5
prnYlgaJpZM4HruZY.gif>

mohit5037, 2016-03-09:
Hi Kris,

I did as you said. I am assuming that it keeps on calibrating the magnetometer in the background so I waved the device for a minute or so and kept on printing the WS_params array using this line of code:

     // Put the Sentral in pass-thru mode
      WS_PassThroughMode();

      // Fetch the WarmStart data from the M24512DFM I2C EEPROM
      readSenParams();

      // Take Sentral out of pass-thru mode and re-start algorithm
      WS_Resume();

Problem is that they don't seems to change at all. Is it because it is not calibrating in the background or there is some other reason.

Here is the video of the movement 
https://youtu.be/jPPG2EwqJsk

Here are the serial logs:
[WS_params_Logs.txt](https://github.com/kriswiner/EM7180_SENtral_sensor_hub/files/165061/WS_params_Logs.txt)

kriswiner, 2016-03-09:
The dynamic calibration occurs continuously and will eventualy iterate to a more or less stable calibration. The calibration parameters will continue to evolve slowly in response to changes in temperature and environment (entering a room with a strong or fluctuating local magnetic field). The asymptoting to a stable calibration will occur faster if you move the device in a figure eight motion for tens of seconds to sample the 3D response surface. Once the calibration is stable there is nothing more for the user to do; the device should give accurate absolute orientation.

The user can go through this procedure every time the device is powered on, or the user can store the calibration parameters using the in-board EEPROM such that every time the device is subsequently powered on after the initial calibration the device can retrieve the calibration parameters and return to the previous state of calibration automatically and at the start of the data stream.

In this sense, the USFS is a calibrate-once, use -forever device. The only reason to change the stored calibration would be if the environment changes substantially. Like if you calibrate it in California in the Summer and then take the device to Chile in the Winter. The average temperature and magnetic field strength of the two environments is quite different and the device would need to be calibrated again. It will calibrate itself eventually but will not start out calibrated upon each power on unless you go through the procedure of calibrating the device and storing the calibration parameters in the EEPROM.

If you calibrate the device and store the parameters in the EEPROM and the device stays in your local area there would be little need to recalibrate and re-store calibration parameters--ever.

I hope this explanation is clear.

jrsmiley, 2017-04-26:
Hi Kris,

I'm having the same problem.  I tried doing the figure eight for about five minutes and the software vs hardware readings never converged.  For a few minutes after calibration, I did manage to get yaw readings that differed by 90 after rotating the sensor 90 degrees with very little drift.  However, after a few more minutes, a 90 degree turn would only show about 60 degrees of difference in yaw and the drift was back in a big way.  

I took pains to do the calibration and measurement in an area relatively free of magnetic interference (my living room - where there are no electronic gadgets other than the laptop I was using to take the measurements).  I'm not sure what I'm doing wrong.

kriswiner, 2019-02-03:
> Hi Kris.
>
> I'm also having the same issue, I fixed the board to a mechanism that can
> rotate only in one axis from 90° to 299°, in the next video I show the
> fixed mechanism which I rotated from 299 to 90°
> https://www.youtube.com/watch?v=u5uOvwxDJqI
>
> But the quaternion output is changing in the three imaginary elements
> giving me a rotation all the angles, the measurement is very consistent
> between repetitions but I won't know what transformation is required
> because all the axis are rotating.
>
> I'm using the ponycube algorithm to display the quaternion rotation and I
> have a rotation very similar to @mohit5037 <https://github.com/mohit5037>
> like if I were spinning the whole board on one corner.
>
> Here I put my output measurements of rotating from 90 to 299° (I have a
> digital angle measurement attached on the mechanism axis)
>
> QUAT angle = [roll, pitch, yaw]
> QUAT value = [w, x, y, z]
>
> For a real measured angle of: 90.0 the output quat was: [0.52, -0.49,
> 0.49, -0.51] and converted to angles: [94.82, 0.57, -89.51]
> For a real measured angle of: 90.58 the output quat was: [0.54, -0.46,
> [... 170 lines cut ...]
> 0.63, 0.35] and converted to angles: [282.27, 38.21, 92.88]
> For a real measured angle of: 299.17 the output quat was: [0.63, 0.28,
> 0.63, 0.36] and converted to angles: [282.12, 35.97, 92.79]
> For a real measured angle of: 299.24 the output quat was: [0.62, 0.3,
> 0.62, 0.37] and converted to angles: [281.9, 33.22, 92.58]

If you are using the EM7180 why not just read the Euler abgles directly
thereby avoiding any errors in conversion. But I suspect the problem like
most others is inadequate calibration.

How do you check your sensors are well calibrated?

EinSoldiatGott, 2019-06-19:
> If you are using the EM7180 why not just read the Euler abgles directly thereby avoiding any errors in conversion. But I suspect the problem like most others is inadequate calibration. How do you check your sensors are well calibrated?
> […](#)

Hi Kris, I'm going to switch the output to Euler angles to release the uC from this process, thanks for pointing me to that approach.

To get the Euler angles, do I have to modify the firmware loaded in the EEPROM?  Or is just enough writing to the register AlgorithmControl   = 0x54 ([2] 1 = HPR output)??

Thanks

kriswiner, 2019-06-19:
You should be able to select either quaternion output or Euler angle output
by changing a bit in a register, no need for new firmware in the EEPROM.