kriswiner/EM7180_SENtral_sensor_hub #6: Missing Documentation

herbertwest3, 2016-03-22:
Marvelous stuff! I have looked at all the documents I could find on pnicorp.com and emdeveloper.com, 
but haven't found the source of the following information about register addresses and meanings
that you have used in your host driver program:

  EM7180_set_integer_param (0x49, 0x00)     

(parameter for "stillness" mode -- what else is available in parameter space?)

and

  writeByte(EM7180_ADDRESS, EM7180_ACC_LPF_BW, 0x03); // 41Hz
  writeByte(EM7180_ADDRESS, EM7180_GYRO_LPF_BW, 0x03); // 41Hz

(are these direct mappings into MPU9250? how did you discover them?)

I have looked at the documents for SENtral-A (for android)  and they hint at all sorts of wonderful
things, but there is no clear mapping to EM7180 equivalents.

Did I miss some documents?

Thanks,

HW

kriswiner, 2016-03-22:
Yes, there are other documents that spell out all of the knobs and user
accessible registers for these kind of things. Generally not open source, so
I will ask you to keep this to yourself. We will be describing some of these
available features in more detail on the github wiki but until then I would
ask that you not share the attached without my express permission. This just
gives you a flavor of everything there.

Kris

<https://github.com/notifications/beacon/AGY1qhhcJFD-mtNb_lz6U0LaDI-rKKC6ks5
pv4izgaJpZM4H10cv.gif>

seamaster101, 2017-08-05:
Hi Kris,
From your documentation I read:
"There are parameters that control how quickly the orientation solution reacts to changes in the acceleration and magnetic field data. The parameters can be set such that the orientation solution reacts very rapidly to changes in acceleration and/or magnetic field, but too much sensitivity can result in unstable or oscillatory orientation solutions. Similarly, a slower response to changes in acceleration or magnetic field effectively results in more averaging of these data which can be useful for avoiding responses to mechanical shocks (when walking, for example) and for detecting and correcting for magnetic anomalies.”
While I know now such parameter exists, I don’t know what to do next. reading this post I realise it might not be public domain...
Is there a way  you can reveal some more information even if it is not in public as in the case from herbertwest3 above
What exactly is this parameter called and  and how do I adjust it? Can you please provide sample?

Thank you for doing all these fun projects possible!
Cheers!
Jordan

kriswiner, 2017-08-05:
Send me an e-mail at tleracorp@gmail.com with your request.