kriswiner/EM7180_SENtral_sensor_hub #8: Confusing Documentation for Ultimate Sensor Fusion board

Wetmelon, 2017-02-24:
I'm having trouble figuring out exactly what's what on the Ultimate Sensor Fusion board.  I'm hooking it up to a Teensy 3.5.

As I understand it...

GND -----> GND
3V3  -----> 3V3
SCL  ------> Teensy SCL
SDA ------> Teensy SDA
INT ------> Teensy GPIO
SDM -----> NC
SCM -----> NC
IO1 ------> NC
IO2 ------> NC
IO4 ------> NC

I thought for a while that SDM/SCM were for the master (i.e. Teensy), but if I understand this image 
[image] correctly, it's actually connected to the sensors (i.e., EM7180 as Master)?

What are IO1, 2, and 4 for?

kriswiner, 2017-02-25:
The connections you have are correct, no need for pullups they are on the board. The master bus is the EM7180 master through which it manages the sensors. The extra EM7180 GPIOs and master bus is exposed in case you want to add other sensors for the EM7180 to manage. Most people do not do this but we do and have and it has been useful at times. You may simply ignore these.

Dealerpriest, 2019-05-29:
It's still not clear how one should treat the pins. I want to solder my USFS to a teensy while being able to use the mentioned pins (SDM, SCM, IO1 IO2, IO4) for other things on the teensy. Can I do so? 
"You may simply ignore these". As in which of the following?
1. It's totally fine to connect them to stuff and they are in high impedance (floating) and would make the USFS react in any way.
2. They have special functionality and will do stuff with the board if conncted and pulled high or low.

I would appreciate to have a clarification. My project needs a lot of pins and I need those spare ones.

Dealerpriest, 2019-05-29:
I have now with rather high confidence verified that I cannot use the pins at the same time as other peripherals (in my use cases I need to connect the spi pins to a nrf24l01+ radio board).

kriswiner, 2019-05-29:
Do not connect the USFS pins that you are not using to the Teensy.