kriswiner/EM7180_SENtral_sensor_hub #9: mpu9250 config file

cardsatgit, 2017-03-13:
Hi Kris,
just quick question as I am facing troubles with firmware generation for em7180 supporting mpu9250 only.
After going over whole documentation that I pulled from EM, registering to forum and sentraltoolkit, I am not able to find mpu9250 on the list after sdk selection. It is there when sdk is not selected. What I am missing here. Appreciate any hint on this.
Regards
Jaroslaw Szulik
cards @ o2 . pl

kriswiner, 2017-03-13:
You would need to generate your own drivers and config file. PNI Corp won't
list it in their tool since they don't have a driver for it. See if the
attached works.

cardsatgit, 2017-03-13:
Thank You for Your time and prompt reply. Did you pm me the attachment or it is somewhere on your github.
Regards

kriswiner, 2017-03-13:
Send me meail at tleracorp@gmail.com, apparently github doesn't like
attachments.

cardsatgit, 2017-03-13:
Thanks, sent.

closed 2017-03-16 by cardsatgit