kriswiner/EM7180_SENtral_sensor_hub #10: How to increase the Sensor Fusion rate as well as the low pass filter bandwiths?

GM223, 2017-04-16:
Hi,
I recently bought the board and the results are pretty amazing.
I am able to get an output data rate of 100hz by using your provided sketch but for my application i require as high a data rate possible.
400hz (which is max hardware fusion rate) of filtered quaternion output should be good enough but here lies the problem.
I have changed various parameters such as :
`writeByte(EM7180_ADDRESS, EM7180_QRateDivisor, 0x01); // same data rate as gyro rate`
` writeByte(EM7180_ADDRESS, EM7180_AccelRate, 0x28); // 400/10 Hz
  writeByte(EM7180_ADDRESS, EM7180_GyroRate, 0x28); // 400/10 Hz`

But the quaternion output rate has not increased over 130 Hz max.
Any help would be appreciated.

Thank you
[image]

kriswiner, 2017-04-16:
Well first thing is I would change this line:

writeByte(EM7180_ADDRESS, EM7180_QRateDivisor, 0x01); // same data rate as
gyro rate

to this:

writeByte(EM7180_ADDRESS, EM7180_QRateDivisor, 0x00); // same data rate as
gyro rate

since the divisor is applied as 1/(1 + qratedivisor) you should have gotten
something like 200 Hz of quaternion output. Not sure if you will get the
full 400 Hz this way, if you need even faster rates you could try to
increase the gyro rate to 1 kHz.

bmegli, 2018-07-04:
@kriswiner 

> since the divisor is applied as 1/(1 + qratedivisor) 

Interestingly you are correct despite that this contradicts documentation in several places:
- Write 0x01 to the QRateDivisor Register. This sets the quaternion output data rate to
equal the GyroRate (EM7180 datasheet, page 22)
- For example, if the GyroRate is set to 200 Hz (0x14) and the QRateDivisor register value is set to
0x04, then the quaternion interrupt rate will be 50 Hz (EM7180 register map, point 4.3)
- The rate at which quaternion interrupts are sent to the host is equal to GyroRate divided by the QRateDivisor register (EM7180 register map, point 4.3)
- and more

I have measured the rate at which I get quaternion interrupts and:
- you are correct ( 1/(1 + qratedivisor )
- EM7180 documentation is wrong (1/qratedivisor)

bmegli, 2018-07-04:
> Not sure if you will get the full 400 Hz this way, if you need even faster rates you could try to increase the gyro rate to 1 kHz.

With USFS I got:
- 333 Hz ODR at 330 Hz Gyro & Accel
- 500 Hz ODR at 500 Hz Gyro & Aceel
- 833 Hz ODR at 1000 Hz Gyro & Accel

Neither of those resulted in setting AlghoritmSlow flag.

ODR was measured as frequency of quaternion interrupts (interrupts that result in quaternion data ready flag)

According to the EM7180 documentation:
- the outputs are updated at a rate limited to the gyro output data rate (ODR), to a maximum of 400 Hz

It remains unclear whether:
- interrupt is generated more often but quaternions updated at 400 Hz
- quaternions are updated at higher rate contradicting documentation

kriswiner, 2018-07-04:
The quaternion rate is the same as the gyro rate guaranteed up to 400 Hz
and maybe higher. The maybe depends on the sensor, the efficiency of the
sensor driver, and your mCU among other things. The fact that you are
seeing a rate as high as 833 Hz is an indication of the kind of limit I
expect. And at that rate you might miss the occasional quat update. I would
run it at 500 Hz in this case if you need the fastest quaternion update.

Not everything your read will be correct.

EinSoldiatGott, 2018-09-01:
Hi.
I have a difference between the rate printed by the sketch and the rate measured with an oscilloscope on the INT pin.

The maximum output I get is approx 60Hz while the sketch says 1khz. I'm using an Arduino UNO with your sketch.

https://photos.app.goo.gl/hST84zsg75DaEPqZ6

Is there any factor in your sketch rate print?

Thanks