kriswiner/EM7180_SENtral_sensor_hub #12: Selecting sensors scale and calibration methods

seamaster101, 2017-07-25:
Hi Kris,
I got my Ultimate fusion sensor board and I got it working right away last night with your example files. I'm still waiting for my Butterfly board to show up. My final configuration will be the ultimate fusion sensor interfaced to butterfly board, but for now I'm using teensy 3.2
Thank you for the examples - that was easy! Now the hard (for me) part... 

Here is the challenges that I'm having with my project. I would appreciate if you can help me with solving the following issues:

1. My goal was to create a roll/pitch compensated compass and select most favorable ranges of the sensors for my application (compass attached to a 50+ feet 25 ton boat, that we can't wave around in figure eight to calibrate) 
Q: what ranges of the accelerometer and gyro I select to get me the best/smoothest output results without too much lag?  Considering the size of the object obviously even in worst conditions, there will be no large values on all of the sensors axis. What are the benefits (if any) choosing lower axis scales for accelerometer and the gyro? 

2. It is challenge to calibrate the compass attached to a vessel (can't wave it around in figure eight) obviously different calibration routine is needed, or is it? No idea how to accomplish that. I read the documentation and understood that files with pre-calibrated parameters can be send to the sensor fusion processor. First I need some method to do the calibration with the sensor mounted to the boat. Here is what I'm thinking could be used:
Case 1: Go to manual calibration mode, drive the boat in a slow circle in calm water over and over maybe for 720deg, keep collecting and evaluating the magnetometer calibration data and if satisfied notify and store data for future warm start.
Case 2: auto-calibration mode where a routine constantly collects calibration data and stores the data for future use if better calibration than the existing one is achieved.
Can you please somehow help with that? I think it will be beneficial to other users as well

3. Output compensated Yaw, Roll, Pitch and Rate of Turn outputs. The yaw, roll and pitch work great now with the hardware fusion, but how do I output compensated gyro ROT in deg/s data? It is ged/sec data of the axis perpendicular to the earth's surface.  I experimented with outputting "gx" variable for test purposes and it "kind of works", but the output is very jumpy and as the variable is from single axis,  I presume it is not accurate at different roll and pitch angles. Any help here will be greatly appreciated!  Is there a variable that fusions and smooths the Z gyro axis?

thank you for your time!

kriswiner, 2017-07-25:
Happy to help you get the USFS working, etc but I am not able to spend the
time to help you answer your project-specific questions pro bono.

Kris

seamaster101, 2017-07-25:
Thank for the to the point answer.  I understand that some of my problems are project specific, but some are general questions. For example selecting ranges and how that impacts the performance of the unit, 
In your opinion, what of the above you can answer pro bono? 
cheers!

kriswiner, 2017-07-25:
All of these questions are application specific; we offer the wiki with
detailed descriptions of what the device can do and how to use it. Plus
there is the data sheet of the MPU9250. I'm not trying to be coy here, I am
very busy with paid work and I do not have the time to help you optimize
performance for your application.

seamaster101, 2017-07-27:
Can you at least help me with how to load lower values  in the gyro and accelerometer? I see that the code is loading +/- 8g for the accelerometer and  +/- 2000dps for the gyro. I would like to test with +/-2g and +/-360dps if possible. Where in the code I change that and do I need to change anything else to have the calculations still be correct?

seamaster101, 2017-07-27:
... and for the record I read all of the wiki and any other documentation that I could find. I "kind of" get it, but I'm afraid not to screw anything up, and then for sure I know, I can't fix it.  
if im correct it should be here:
 //Write desired sensor full scale ranges to the EM7180
    EM7180_set_mag_acc_FS (0x3E8, 0x08); // 1000 uT, 8 g
    EM7180_set_gyro_FS (0x7D0); // 2000 dps
  but not sure what the corresponding values for my desired parameters are...

kriswiner, 2017-07-27:
It's done here:

//Write desired sensor full scale ranges to the EM7180
  EM7180_set_mag_acc_FS (0x3E8, 0x08); // 1000 uT, 8 g
  EM7180_set_gyro_FS (0x7D0); // 2000 dps

in the pratemer transfer sectionin setup.

Even though the sensor FS range changes, the EM7180 output is always at a
fixed FS range.

kriswiner, 2017-07-27:
accel can be 0x02, 0x04, 0x08 or 0x10, that is 2, 4, 8, or 16 g full scale.

seamaster101, 2017-07-27:
thank you. I will try it.

seamaster101, 2017-07-27:
Do you have the values for the gyro by any chance too?

closed 2017-07-27 by seamaster101

kriswiner, 2017-07-27:
> 5sR_o5gaJpZM4OjIOm>.

Datasheet

250, 500, 1000, 2000

seamaster101, 2017-07-27:
which data sheet? i look into 9250 and into 7180. Sorry I couldn’t find that data
I’m missing it?

kriswiner, 2017-07-27:
> 08gaJpZM4OjIOm>.

Really?

https://www.invensense.com/products/motion-tracking/9-axis/mpu-9250/

seamaster101, 2017-07-27:
LOL,
‘got that far myself..

seamaster101, 2017-07-27:
Thank you for putting up with me…
Maybe I didn’t explain my question. i see the values, but thats not what I was after. I think i figured it out finally 
250, 500, 1000, 2000 is 0xFA, 0x1F4, 0x3E8 and 0x7D0
It took me awhile, but i got it.
thanks!