kriswiner/EM7180_SENtral_sensor_hub #14: USFB code for Butterfly

seamaster101, 2017-07-28:
Hi Kris,
I have now the USFB and the Butterfly board.
I manged to get the USFB do what I want with Teensy3.3 board. using the code from here:
https://github.com/kriswiner/EM7180_SENtral_sensor_hub/blob/master/WarmStartandAccelCal/EM71280_MPU9250_BMP280_M24512DFC_WS_Acc_Cal.ino
Now I would like to get the code to run on Butterfly.  
I can compile the samples for the butterfly board and upload them successfully. So I think all of the per-requisites are there...
One challenge I believe is that Butterfly board does not get recognition from arduino.h. The second challenge is that the code for  USFB uses "i2c_t3.h" which is teensy specific (optimized).
If I try to compile the EM71280_MPU9250_BMP280_M24512DFC_WS_Acc_Cal.ino on the Butterfly board I get ""wire" was not declared in this scope" which is expected .
Considering that both boards that I'm trying to use are yours, can you help with the integration?

kriswiner, 2017-07-28:
> WarmStartandAccelCal/EM71280_MPU9250_BMP280_M24512DFC_WS_Acc_Cal.ino

I sent you a Butterfly sketch for the USFS with the appropriate wire calls.
Can;t you use this? I cannot write code for you...

seamaster101, 2017-07-28:
Kris,
I do not expect you to write code for me. All I expect is samples that I can bastardize and write (read copy and paste with very little writing) my version. I don't think this is unreasonable expectation on my part. 
Sorry I could not recall if you sent me code with "wire" calls. I used Teensy till now so I don't think I ever needed wire appropriate code. Maybe I'm mistaken as I have been working too long through the nights. If you send it to me again or remind me where to find it I will be most most likely sufficient and greatly appreciated.

seamaster101, 2017-07-29:
I did find code posted in your examples that uses wire.h., then modified the code and it works now.  I have other issue with board compatibility, but I will open another issue. This one is closed now.

closed 2017-07-29 by seamaster101