kriswiner/EM7180_SENtral_sensor_hub #15: Yaw Calibration

DustinLehmann, 2017-08-12:
Hi,
somehow the Yaw/Mag Calibration is not working for me. I'm using the EM71280_MPU9250_BMP280_M24512DFC_WS_Acc_Cal.ino Sketch. I'm waving the device in an figure 8 motion, and after some time when I try to move the Sensor by 90 degrees it works fine. But if I move the sensor around a bit and placing it back where it came from, the Yaw reading differs a lot. Usually about 10-20 degrees. Furthermore, even if i enter the Magnetic declination for my city it never shows the real north. If i reran the whole process I will get the 90 degrees offset for turns, but the same direction in my room is now off about 60 degrees or so. Am I doing something wrong?

Best Regards
D. Lehmann

kriswiner, 2017-08-12:
Where are you calculating Yaw? Are you taking it from the EM7180 directly
or are you converting to YPR from the EM7180 quaternions? I suspect the
latter, and you are possibly not using the quats in the right order or
something. Try taking the Euler angles directly from the EM7180.

DustinLehmann, 2017-08-12:
I'm using the yaw calculation from Quaternions from your sketch, I did not change it. How do I access the Euler-Angles from the EM7180? Do I have to do it by changing the Algorithm control register (0x54)? If I read the "magnetometer calibration completed" bit in the algorithmStatus-register it is set to 1, but the yaw still drifts a lot if I set it back to the initial position. But the sensor reports that the calibration is completed after a few seconds of waving in figure 8 motion. 

I have a second question: You are storing the Acceleration calibration data in the 0x5b register, which is not specified in the documentation, so I guess it's a user selectable register? Do you use the data in it by using this register in the configuration file or where is the acceleration config data used?

DustinLehmann, 2017-08-13:
I did some additional research today, with the hardware euler angles:
- sometimes it gets close to the real heading, but after 5-6 minutes it drifts away for about 20 deg 
- if I save the WS-data and restore it, without moving the sensor, most of the time the angles before and after reboot and using the WS data differ for more than 90 deg
- sometimes the hardware yaw starts drifting continiously without moving the sensor after a period of constant heading 
- pitch and roll are good, stable and accurate

Could the magnetometer be defect? I tried hooking up a different one but my registration for the SENtral toolkit have Not been validated or answered...

Do you have some Ideas what I could try to get a stable and (mostly) accurate yaw angle?

Best Regards

kriswiner, 2017-08-13:
This behavior you are describing isn't normal nor do we see this in our
applications. Please send me an e-mail at tleracorp@gmail.com so we can
help you diagnose the problem(s).

Kris

Patrickyp, 2018-06-09:
Hi kris,

If using not using passthrough mode, you can do the figure 8 anytime the sketch is outputting data to calibrate, while passthrough mode the calibration is done at the beginning before data collecting start am I correct?  

Do you recommend no passthrough or passthrough mode?

kriswiner, 2018-06-09:
You can use the SENtral's own calibration in normal (not passthrough) mode
and achieve a typical 2 degree rms heading accuracy,so this is what I would
recommend. I would also recommend use of warm start by which calibration
parameters are saved and the SENtral started with previously saved
parameters on subsequent power on. This way, you don't have to do the
figure eight each time you use it.