kriswiner/EM7180_SENtral_sensor_hub #16: EM7180 Nano Board

ReneKaminski, 2017-10-23:
Hello Kris,
Sorry that I contact you this way, but the Facebook app on your Tindie account isn't working for me.
Me and another project group at my university are working on a project concerning orientation. We tried the BNO055 but are not getting reasonable results out of this thing. 
We read your article about the EM7180 but couldn't find the Nano board in your Tindie Shop. Are you still selling them?

kriswiner, 2017-10-23:
Hi Rene,

You can reach me at tleracorp@gmail.com.

I never put the EM7180 Nano board design up for sale at Tindie, I just
designed and tested a couple to prove feasibility.

We currently sell only the USFS version you can see here:

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution

Is smaller size a requirement for your application

Kris?

ReneKaminski, 2017-10-23:
Hello Mr. Winer (?),
many thanks for the prompt reply. The MPU9250 s have to be woven/glued into/onto the fingers of a glove and also on the back of the hand. This will become the remote for a robot hand. In my case it´s going onto shoulder and arm. While the bigger breakout would work for me, the smaller one would be definitely more sophisticated for the glove.
Kind regards,
René

kriswiner, 2017-10-23:
Well, if you are careful, you could just trim 1-mm from each edge and cut
the size down that way. But I have no plans to produce the Nano version for
sale at Tindie.

If you wanted to buy 100 of them, on the other hand, I could have them
produced in China, but this would take six weeks or so.

ReneKaminski, 2017-10-25:
Sorry for my delayed answer. I discussed your offering with my colleauges. We appreciate your offer very much, but we can't wait 6 weeks because of our deadline.
Kind regards,
René

kriswiner, 2017-10-25:
OK Rene,

Perhaps the USFS
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/?pt=full_prod_search>
would do? These are available now and I could offer you a 10% discount for
an order of 10 or more. I mean order nine get one free!

Anyway, let me know if I can help.

Kris

ReneKaminski, 2017-10-27:
Hello Kris,
do you know the delivery time from your Tindie Shop to Germany by chance?
Kind regards,
René

kriswiner, 2017-10-27:
Hi Rene,

Let's switch further discussion to regular e-mail. I am at
tleracorp@gmail.com. If I get the package in the mail today (Friday) you
would have them next week.

Kris