kriswiner/EM7180_SENtral_sensor_hub #17: Values not stable on Warm Start

afnan, 2017-11-16:
Hi,
Hope you are fine. I found your board very well designed and easy to work with. However I am unable to get stable readings on warm start. Everything goes well on No Warm start and calibration goes well and i press 1 to save settings. However after rebot it seems that it does not remember any setting and goes out of whack. Readings for both Yaw, Pitch and Roll would keep on changing with more than 10Deg difference. 

I am using ESP8266 and only change I made was I excluded `#include <i2c_t3.h>` and used` #include <Wire.h> ` and modified similar things which were causing error in regards to this. 

Using this https://github.com/kriswiner/EM7180_SENtral_sensor_hub/tree/master/WarmStart
Thanks

kriswiner, 2017-11-16:
> tree/master/WarmStart

Hi Afnan,

Likely you don't have the latest warm start sketch. I will ask Greg to see
if he can help here.

Greg, can you give Afnan the latest USFS sketch with warm start capability
lease?

Kris

kriswiner, 2017-11-17:
Hi Afnan,

 

Give this one a try; it should work with ESP82XX as well with the right I2C library/call…

 

Best,

Greg

 

From: Kris Winer [mailto:tleracorp@gmail.com] 
Sent: Thursday, November 16, 2017 9:33 AM
To: kriswiner/EM7180_SENtral_sensor_hub
Cc: Greg Tomasch
Subject: Re: [kriswiner/EM7180_SENtral_sensor_hub] Values not stable on Warm Start (#17)

 

Hi Afnan,

 

Likely you don't have the latest warm start sketch. I will ask Greg to see if he can help here.

Greg, can you give Afnan the latest USFS sketch with warm start capability lease?

Kris

afnan, 2017-11-18:
Probably above message was incomplete I cannot find any link  or info. Thanks

kriswiner, 2017-11-18:
I see the attachment, do you not?

kriswiner, 2017-11-18:
send me an e-mail at tleracorp@gmail.com then.

afnan, 2017-11-18:
Email sent

afnan, 2017-11-19:
Thank you for sending the new code. It works fine, however I have seen that frequency of fail startup has increased. Do you know what is the reason for following ? any info on helping understand would be great. Thanks

`Scanning...
I2C device found at address 0x28  !
done

EM7180 ROM Version: 0xE69
Should be: 0xE609
EM7180 RAM Version: 0x17435
EM7180 ProductID: 0x80 Should be: 0x80
EM7180 RevisionID: 0x2 Should be: 0x02
A barometer is installed
A temperature sensor is installed
EEPROM detected on the sensor bus!
EEPROM uploaded config file!
EEPROM upload successful!
Acceleromater Calibration Data:
X-acc max: -1
Y-acc max: -1
Z-acc max: -1
X-acc min: -1
Y-acc min: -1
Z-acc min: -1
***No Accel Cal corrections***
!!!Warm Start corrections applied!!!
Beginning Parameter Adjustments
Magnetometer Default Full Scale Range: +/-1000uT
Accelerometer Default Full Scale Range: +/-8g
Gyroscope Default Full Scale Range: +/-2000dps

Soft WDT reset

ctx: cont 
sp: 3fff0470 end: 3fff0780 offset: 01b0

>>>stack>>>
3fff0620:  3ffef616 00000000 00000000 3ffef614  
3fff0630:  3ffef616 00000140 00000006 4020120c  
3fff0640:  00000002 00000001 00000004 40201300  
3fff0650:  00000028 00000001 3ffef616 3ffef730  
3fff0660:  00000000 00000054 3ffef611 3ffef730  
3fff0670:  000000cb 00000064 3ffef611 40206fa8  
3fff0680:  00000000 00000054 3ffef600 40206fd0  
3fff0690:  3ffef4e8 3ffef4ec 000000bb 4020341e  
3fff06a0:  3ffef4e8 3ffef4ec 000007d0 402036cf  
3fff06b0:  3ffef4e8 3ffef4ec 3ffef4f0 402064f0  
3fff06c0:  feefeffe feefeffe feefeffe feefeffe  
3fff06d0:  feefeffe feefeffe feefeffe feefeffe  
3fff06e0:  feefeffe feefeffe feefeffe feefeffe  
3fff06f0:  feefeffe feefeffe feefeffe feefeffe  
3fff0700:  feefeffe feefeffe feefeffe feefeffe  
3fff0710:  feefeffe feefeffe feefeffe feefeffe  
3fff0720:  feefeffe feefeffe feefeffe feefeffe  
3fff0730:  feefeffe feefeffe feefeffe feefeffe  
3fff0740:  feefeffe feefeffe feefeffe feefeffe  
3fff0750:  feefeffe feefeffe feefeffe 3ffef754  
3fff0760:  3fffdad0 00000000 3ffef74c 40207768  
3fff0770:  feefeffe feefeffe 3ffef760 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset`

afnan, 2017-11-19:
Sometimes it would throw error as follow 

`Scanning...
Unknow error at address 0x01
Unknow error at address 0x02
Unknow error at address 0x03
Unknow error at address 0x04
Unknow error at address 0x05
Unknow error at address 0x06
Unknow error at address 0x07
Unknow error at address 0x08
Unknow error at address 0x09
Unknow error at address 0x0A
Unknow error at address 0x0B
Unknow error at address 0x0C
Unknow error at address 0x0D
Unknow error at address 0x0E
Unknow error at address 0x0F
Unknow error at address 0x10
Unknow error at address 0x11
Unknow error at address 0x12
Unknow error at address 0x13
Unknow error at address 0x14
Unknow error at address 0x15
Unknow error at address 0x16
Unknow error at address 0x17
Unknow error at address 0x18
Unknow error at address 0x19
Unknow error at address 0x1A
Unknow error at address 0x1B
Unknow error at address 0x1C
Unknow error at address 0x1D
Unknow error at address 0x1E
Unknow error at address 0x1F
Unknow error at address 0x20
Unknow error at address 0x21
Unknow error at address 0x22
Unknow error at address 0x23
Unknow error at address 0x24
Unknow error at address 0x25
Unknow error at address 0x26
Unknow error at address 0x27
Unknow error at address 0x28
Unknow error at address 0x29
Unknow error at address 0x2A
Unknow error at address 0x2B
Unknow error at address 0x2C
Unknow error at address 0x2D
Unknow error at address 0x2E
Unknow error at address 0x2F
Unknow error at address 0x30
Unknow error at address 0x31
Unknow error at address 0x32
Unknow error at address 0x33
Unknow error at address 0x34
Unknow error at address 0x35
Unknow error at address 0x36
Unknow error at address 0x37
Unknow error at address 0x38
Unknow error at address 0x39
Unknow error at address 0x3A
Unknow error at address 0x3B
Unknow error at address 0x3C
Unknow error at address 0x3D
Unknow error at address 0x3E
Unknow error at address 0x3F
Unknow error at address 0x40
Unknow error at address 0x41
Unknow error at address 0x42
Unknow error at address 0x43
Unknow error at address 0x44
Unknow error at address 0x45
Unknow error at address 0x46
Unknow error at address 0x47
Unknow error at address 0x48
Unknow error at address 0x49
Unknow error at address 0x4A
Unknow error at address 0x4B
Unknow error at address 0x4C
Unknow error at address 0x4D
Unknow error at address 0x4E
Unknow error at address 0x4F
Unknow error at address 0x50
Unknow error at address 0x51
Unknow error at address 0x52
Unknow error at address 0x53
Unknow error at address 0x54
Unknow error at address 0x55
Unknow error at address 0x56
Unknow error at address 0x57
Unknow error at address 0x58
Unknow error at address 0x59
Unknow error at address 0x5A
Unknow error at address 0x5B
Unknow error at address 0x5C
Unknow error at address 0x5D
Unknow error at address 0x5E
Unknow error at address 0x5F
Unknow error at address 0x60
Unknow error at address 0x61
Unknow error at address 0x62
Unknow error at address 0x63
Unknow error at address 0x64
Unknow error at address 0x65
Unknow error at address 0x66
Unknow error at address 0x67
Unknow error at address 0x68
Unknow error at address 0x69
Unknow error at address 0x6A
Unknow error at address 0x6B
Unknow error at address 0x6C
Unknow error at address 0x6D
Unknow error at address 0x6E
Unknow error at address 0x6F
Unknow error at address 0x70
Unknow error at address 0x71
Unknow error at address 0x72
Unknow error at address 0x73
Unknow error at address 0x74
Unknow error at address 0x75
Unknow error at address 0x76
Unknow error at address 0x77
Unknow error at address 0x78
Unknow error at address 0x79
Unknow error at address 0x7A
Unknow error at address 0x7B
Unknow error at address 0x7C
Unknow error at address 0x7D
Unknow error at address 0x7E
No I2C devices found

EM7180 ROM Version: 0xFFFF
Should be: 0xE609
EM7180 RAM Version: 0x255255
EM7180 ProductID: 0xFF Should be: 0x80
EM7180 RevisionID: 0xFF Should be: 0x02
A barometer is installed
A humidity sensor is installed
A temperature sensor is installed
A custom sensor is installed
A second custom sensor is installed
A third custom sensor is installed
EEPROM detected on the sensor bus!
EEPROM uploaded config file!
EEPROM CRC incorrect!
EM7180 in initialized state!
No EEPROM detected!
`

kriswiner, 2017-11-19:
The first error is common if there is not enough power available for the
ESP8266. The second if there is a bad connection on the I2C lines, like no
3V3 power to the board.

I suspect in both cases, your connections are not robust.

afnan, 2017-11-20:
Well everything works fine on first startup i.e when plugged into laptop. those two errors occur when i press reset button on node mcu. Probably they take some time in restarting. First error happens if i put i2c scan routine after connection to wifi hotspot and second occurs if I place it in very start of setup function. Probably it has something to do with power rise or something.

kriswiner, 2017-11-20:
Perhaps it is the ESP8266 board you are using?