kriswiner/EM7180_SENtral_sensor_hub #18: Sentral Toolkit

DustinLehmann, 2017-12-11:
Hello,
do you know what happened to the Sentral Toolkit on www.sentraltoolkit.com? I cannot reach it anymore.

Best Regards
Dustin Lehmann

ReneKaminski, 2018-01-15:
@DustinLehmann Hi, do you know something more now? I wrote the support last week but no answer so far.

Best regards,
René

kriswiner, 2018-01-15:
As far as I know EM Microelectronics and PNI Corp have removed the SENtral
toolkit from public view since they have decided no longer to provide
low-volume customer support. If you have specific questions about the
EM7180, or the USFS
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/>with
embedded EM7180, or wish to discuss custom work with the EM7180 please let
me know at tleracorp@gmail.com.

ReneKaminski, 2018-01-15:
Dear Kris,

I had to buy the EM7180 from a german distributor..  I had really liked to buy some of your assembled breakoutboards though.

The damn bureaucracy in my university department.

 

If I understand it right there is no way to use the device without config file. In the data sheet PNI corp still refers to the homepage and they don´t answer to email.

 

Is there a way to make this file without the homepage? 

Or would you provide me the file you have used, please??       (I´ll use MPU9250 too, but don´t need an altimeter)

 

I´m really sorry bothering you once more.

 

Thank you very much,

Kind Regards,

René

Gesendet: Montag, 15. Januar 2018 um 18:46 Uhr

crispc85, 2018-04-11:
Any word on if the Sentral Toolkit can still be acquired by end-users? I'd really like to be able to make a custom configuration file for my application (mainly just need to be able to change the axis orientations).

Please help!

Thanks!

kriswiner, 2018-04-11:
Don't think so...PNICorp removed it a year ago or so.

crispc85, 2018-04-11:
Ok...the Wiki page should really be updated to say that the device configuration tool is no longer available, as it kind of renders the product useless to consumers then unless the default configuration is all that is needed. I never would have ordered these for my project if I'd known they weren't configurable any longer.

kriswiner, 2018-04-11:
The MPU9250 was never supported by the configuration tool anyway, wouldn't
have done you too much good.

Not sure why you need to change anything, the USFS you bought from us have
the correct firmware for the boards and should work as advertised.

gerowtj, 2022-01-07:
Hi Kris.
Sorry to drag up this topic again.  I have the EM7180 USFS Sentral with the LSM6DSM, LIS2MDL and LPS22HS sensors.  My project coordinate system has the front off the project box as North, the Right side as East and the Bottom as Down.
The EM7180 is mounted with the Sentral North facing the Left side of the project box (West) and the Sentral East facing the front of the project box (North).   Sentral Down matches the project box Down.  

I'm assuming this will require the Sentral Configuration File to be modified in order to get the Sentral NED to match my project coordinate system.  Is this correct?  Any other way in software to make the correction since the Configuration Tool is no longer available?

Worst case I can attempt to redesign the circuit board so the Sentral coordinate system and project coordinates match, but that's not as desirable at the moment.

Thanks,
Tom

kriswiner, 2022-01-07:
This should be a simple matrix operation to convert the rotation matrix to
your coordinates. I will ask my colleague Greg if he has an example.

Can you help Tom with this rotation problem? Thanks...