kriswiner/EM7180_SENtral_sensor_hub #19: No EEPROM detected

damarist, 2017-12-19:
Dear Kris,

This is the second board I got from you that gives me this problem. The first one I got to use a couple times before this happened, the second one I had barely bought it and it gave me the same problem on the first use. On your documentation, it mentions there's no need for uploading the configuration file again. I'm using with a Teensy LC. 3V come from Teensy to Sensor Fusion. 

In an attempt to recover the file I got the SDcard to SPI adapter board you used from Adafruit and ran the Firmware.ino just changing the register address for MPU9250 instead of the BMX055. But I don't have the .fw configuration file for this new board of yours so it's not storing to EEPROM. The SENtral toolkit website you mention has been down for months already. Could you upload the xxx.fw file so I can reach a solution?
[image]

damarist, 2017-12-19:
Ok, now I ran EM7180_MPU9250_BMP280_t3 Basic Example Code instead. But it gets stuck at scanning and doesn't show me what is connected to the bus, this is the window I get and it's been for almost 10 minutes the same.
[image]

kriswiner, 2017-12-19:
Looks like there is a problem with your I2C bus, it is mot detecting any
I2C devices. Are you sure you have the correct connections?

kriswiner, 2017-12-19:
There is no need to write or over write the firmware on the eeprom; each
usfs is loaded with the correct firmware and thoroughly tested for proper
function before shipping. Once you fix the i2 c bus problem the device will
work.

damarist, 2017-12-19:
It was working fine, suddenly it gave me the no EEPROM detected on Sunday night so I tried my new spare one on and it was the same. I have connected SCL to pin 16 and SDA to 17, Vcc is using the 3V output on Teensy and all grounds are connected. My INT pin I moved to 11 because I'm trying to implement wake-on-motion from Low Leakage Sleep mode and I need a pin with LLWU capability. But even if I connect to 8, nothing happens. It gives me the same message. I wish I had a snapshot of before when it was working, it was giving the quaternions data I need. All I keep getting is 
[image]

kriswiner, 2017-12-19:
Your i2c isnt working. Can you verify with some other i2c sensor?

Patrickyp, 2018-06-13:
I am getting the same problem where it gets stuck at scanning sometimes.  I would have to replug the usb cable from teensy to laptop and it seems to fix it.  But the problem at times will show up again although most of the time it works.

kriswiner, 2018-06-13:
This sounds like a USB cable/connector problem to me.

Patrickyp, 2018-06-15:
I changed the usb cable and seems to have same problem.  Any chance it can be the connections with the pins?  I am using a cheap breadboard.