kriswiner/EM7180_SENtral_sensor_hub #22: changing reference frame of USFS outputs

rifotu, 2018-02-21:
Hello,

I bought a USFS last week and waiting for it to arrive.  In the mean time, I am trying to make some preperations. 

For my application, I think I need to change the reference frame of my USFS so that rotations produce a single axis change. 

Currently I use a BNO055 board (not satisfied with the calibration issues, hence I bought a USFS) and try to solve the below problem. When my USFS arrives, I'll replace BNO055.

In the below video, when I place the IMU on the table and rotate it in a single axis, I only get a change in X axis (which is what I expect)

https://vimeo.com/256816666

I then place the IMU upright and rotate it downwards (as shown in video, 0:26) orientation values change in all directions. 
Now I am trying to change the reference frame of my IMU such that the rotation at 0:26 causes a change in a single axis.

I tried to post process (to no avail) the  incoming orientation values so that the values at 0:26 (219.8, -1.63, 90) become (0, 0, 0)
I thought about using a rotation matrix but I don't know how to generate input vectors from orientation

I think I will need to do this with USFS too so I would be happy if you can point me about how to do this?

any help would be appreciated
thanks
rifo

kriswiner, 2018-02-21:
Hi Rifo,

I am not sure exactly what you are proposing. When the sensors are well
calibrated you should be getting absolute orientation from the SENtral.
Rotating the yaw pitch and roll into some other frame is possible in
general but i can;t help you with your specific problem. Perhaps Greg has a
link to some useful information in this regard.

Kris

rifotu, 2018-02-21:
Hi Kris,

My aim is to find the rotation angle during the 0:26 movement (video) To me this looks like a single axos rotation but IMU output changes in all dimensions.

It is easy to find the rotation angle for the first movement in the video but I cant calculate the angle for the latter movement

If there is an other way to find this rotation angle from the IMU output I would be happy to work on it