kriswiner/EM7180_SENtral_sensor_hub #24: Acceleration Data Extraction

afnan, 2018-05-21:
Hi,

I am using sensor fusion board with ESP8266. Everything works exceptionally well. However, I am interested to get acceleration value along with yaw, pitch, and roll. When I try to put the sensor in passthrough mode reading from yaw, pitch and roll become zero and I cannot get ax,ay and az other than zero.

Question is that is it possible to use both modes together? if not any other way out to get acceleration? maybe through Quaternions?

kriswiner, 2018-05-21:
You can read scaled accel, gyro, and mag data in normal mode while also
getting hardware yaw, pitch, and roll, etc. Not sure what else you are
trying to do here...

afnan, 2018-05-21:
I believe the main logic you suggest above is same as mine (in terms of acquiring Acceleration data). I got zero values due to aRes variable being zero in my equation. Before diagnosing the this problem I thought values are zero so I should try passthrough mode. 

Now I am getting values in normal mode, not passthrough mode. 

So as values are Scaled as you said, removing aRes from my equation but applying `accelBias` is correct? 

Thanks

kriswiner, 2018-05-21:
You are mixing passthrough and normal mode methods, so no this is not
correct, aRes and accelBias are not used in normal mode.

afnan, 2018-05-21:
Just to understand a bit more. When does acceleration correction from eeprom is applied to acceleration reading? in your code above what is `0.000488`

`x = (float)accelCount[0]*0.000488;  // get actual g value
    ay = (float)accelCount[1]*0.000488;
    az = (float)accelCount[2]*0.000488;
`

kriswiner, 2018-05-21:
0.000488 is how the Sentral accel data is converted to g's, the scale
factor. It is fixed for all settings of the full range. This is one example
how normal mode differs from passthrough mode.

afnan, 2018-05-21:
awesome. Thanks