kriswiner/EM7180_SENtral_sensor_hub #26: Incorrect readings for mbar and altitude

Decman84, 2018-05-29:
Hi Kris,
I’m using the teensy 3.5 and the e7180 and using this code while most output looks good my altitude based reading are off by quite a bit... aka I’m at approx 190m (according to google) yet my sensor output is suggesting -50 consistently.
It appears to be sensing variation at this level it’s just off by quite a factor. Can you advise if you think the sensor is faulty or whether I just need to baseline the config somehow

Decman84, 2018-05-29:
Hi Kris,
Thanks for the speedy response. I was just looking at the output from your standard code set.
Do I need to do further calculations on top of that output?

By the way fantastic sensor, it’s early days for me but loving your work.

Declan

The barometer is an absolute pressure sensor not an altitude sensor. it is
common for absolute altitude estimation to be off by quite a lot (<20%)
since it varies along with the local weather. The altitude estimation is,
however, pretty good for relative changes.

But I will admit, a 250 meter discrepancy is much larger than anything I
have seen. I am at 330 meters and often see a reading as low as 250 m or so
and as high as 350 m.

Are you sure you have the units right? The formula is for specific units,
you can look it up online.

kriswiner, 2018-05-29:
> kriswiner/EM7180_SENtral_sensor_hub/issues/26#issuecomment-392967276>, or

No output should be in feet I think. Still -50 feet is a lot different from
+400 feet...

Decman84, 2018-05-29:
Ahh yes, slight typo in my case.
Not a critical sensor for my use case at the moment so the variation is not that important.
At some stage ill get another em7180 as my prototype progresses and that may shed light on where the issue lies.
Thanks for your help.
Declan