kriswiner/EM7180_SENtral_sensor_hub #27: Images absolute heading accuracy

kriswiner, 2018-06-27:
[image]

kriswiner, 2018-06-27:
[image]

kriswiner, 2018-06-27:
[image]

kriswiner, 2018-06-27:
[image]

kriswiner, 2018-06-27:
[image]

kriswiner, 2018-06-27:
[image]

kriswiner, 2018-06-27:
[image]

kriswiner, 2018-06-27:
[image]

kriswiner, 2018-06-27:
[image]

RyAndrew, 2019-02-24:
Here's another close up photo.
It has a bit more contrast than the photo found on this page:
https://github.com/kriswiner/EM7180_SENtral_sensor_hub/tree/master/EM7180_LSM6DSM_LIS2MDL_LPS22HB_Butterfly
@kriswiner feel free to use it if you want
[image]

kriswiner, 2019-02-24:
Nice image and thanks, but Tindie is complaining the image isn't tall
enough (by pixels...needs to be 866 or more). Tindie is picky that way I
guess...

RyAndrew, 2019-02-24:
Here's a bigger one!
[image]

kriswiner, 2019-02-24:
Thanks, I can see every nook and cranny now!

RyAndrew, 2019-02-24:
Drat. I see you have a photo of the back side as well. I already soldered the pins onto the first, but luckily I had a second. here is a similar dimensioned photo of the back:
[image]