kriswiner/EM7180_SENtral_sensor_hub #29: Warmstart seems not complete

arduz, 2018-07-02:
Hi, I have calibrated my Sentral successful and stored the calibration values in EEPROM. Then I have used your sketch for Warmstart (the newest Version) and used it without input from Serial Monitor (I have set warm_start = 1, always). I have mounted the Sentral on a breadboard and print out the values for yaw, pitch and roll on a little TFT screen. But the values (especially for Yaw) after restart (Warmstart) are horrible. However I move the board with Sentral three times in a 8 figure and then check the values they are wonderful (differences less than 2 degrees). Have you an explanation therefore? I have expected, that the values after Warmstart are ok without necessity to execute the 8-figure-motion.

kriswiner, 2018-07-02:
Looks like you are not storing the warm start parameters properly. What is
your MCU?

arduz, 2018-07-03:
Hello,
my MCU is Teensy 3.5. The warm-start parameters I have stored with the sketch EM71280_MPU9250_BMP280_M24512DFC_WS_Acc_Cal.ino. I"ll try to repeat the calibration procedure described in the users guide chapter F with this sketch. Is it possible, that the stored parameters in the EEPROM was deleted because I used the Sentral (EM7180) in a full sunny environment? Should I be more careful with sun light?

kriswiner, 2018-07-03:
No, sunlight will not affect the EEPROM unless exposed over many weeks.

 You can check that you can read the warm start parameters and make sure
they make sense (not all 0xFF, etc).

arduz, 2018-07-04:
I assume, that I can use e.g. the function "readSenParams()" from the calibration sketch and print out the params on Serial Monitor. Or is there a better possibility to check the validity of the stored Parameters?

braincore, 2018-08-07:
@arduz: Be sure to check that the sensor is not reporting a magnetic anomaly. If so, the magnetometer is ignored and convergence to heading is poor. Sometimes, waving the sensor around gets rid of the magnetic anomaly report and then convergence to yaw happens quickly. I've observed this behavior even with warm start parameters extracted in the same location without a magnetic transience warning.