kriswiner/EM7180_SENtral_sensor_hub #31: Increasing orientation convergence speed

braincore, 2018-07-29:
In the documentation ("G. Tuning the SENtral Sensor Fusion Algorithms"), you mention that "There are parameters that control how quickly the orientation solution reacts to changes in the acceleration and magnetic field data." I do not see these parameters in the EM7180 datasheet and I don't see any indications of them in your example sketches. Any suggestions where I can find these parameters? Are they part of the SENtral config tool?

kriswiner, 2018-07-29:
Send me e-mail at tleracorp@gmail.com and I will send you a document with
more details.

closed 2018-07-29 by braincore

afnan, 2018-07-29:
Isnt it the delay parameter ? @kriswiner I wonder how your computer file organization looks like when you work on too many projects and you still have all the files :) whereas, I am still trying to be an organized person

kriswiner, 2018-07-30:
I don't understand your question.

afnan, 2018-07-30:
@kriswiner in regards to the reaction settings I believe it was the update rate we set in the main. On the other hand I said I was thinking since you work on many projects and have many products out in the market how do you manage all the information on your computer and how do you work on these products in parallel.

kriswiner, 2018-07-30:
For best results I would set the accel/gyro sample rates to 1 kHz and mag
data rate to 100 Hz and then the quaternion rate to 200 or 400 Hz. I would
also use 1 MHz for the I2C bus speed if your MCU can support this.

As far as organization and management, nothing special. File folders on a
Windows machine and github repositories.

And I only manage products I use and am interested in myself. In other
words, I know my products.

What other way is there to manage a business?