kriswiner/EM7180_SENtral_sensor_hub #33: Are you guys ever gonna release a product with the proprietary sensor fusion you talk about in the wiki?

IndianBoy42, 2018-08-12:
The one you talk about in here: https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/K.-Limits-of-Absolute-Heading-Accuracy-Using-Inexpensive-MEMS-Sensors

I'm working on a robotics project/application that needs really good heading accuracy (<=1deg) and responsiveness (>250Hz) at moderately high velocities (10m/s).  Right now it seems the EM7180 USFS is the closest product (that doesnt cost 1000s of dollars), but apparently your algorithm is even better.

Do you have plans to release a product/module with this? Or could I talk to you about privately producing a few modules for me. 

Thank you

IndianBoy42, 2018-08-12:
Also, do you have experience with the more expensive IMU AHRS solutions like the Redshift UM7, Fairchild FMT1030, Xsens Mti etc? Any products in the 100-200usd range that provide good performance out of the box?

kriswiner, 2018-08-12:
From our testing and that reported by others no available solution in the <
$10K category does better than our inexpensive USFS solution.

We are working to build a similar device that uses our proprietary
calibration and fusion algorithms, but we are a few months away from
general availability. You can keep track of our progress here
<https://hackaday.io/project/160283-max32660-motion-co-processor>.

In the meantime, I would recommend that you try the EM7180 + ST
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm--lis2md/>
sensors solution. If you were to buy five, say, then after proper
calibration each would produce < 2 degree rms heading accuracy and two or
three of them would produce ~ 1 degree rms heading accuracy. At least this
way you could do basic testing to determine whether 1 degree heading
accuraxy is sufficient in your application.