kriswiner/EM7180_SENtral_sensor_hub #34: Why search for sensors before of the initializacion?

EinSoldiatGott, 2018-08-15:
Hi, I have some boards from tindie with the MPU9250.

I'm migrating your code to C to be used on one CCS26xx

I'm in the init function. I have correct communication but  I cannot see the MPU9250. It only finds the BME280.

This is my output:

> #000002 [ 0.011 ] INFO: (project_zero.c:574) Initialize i2c
> #000003 [ 0.013 ] INFO: (EM7180.c:50) ROM Ver: 58889 Should be 58889 OK
> #000004 [ 0.013 ] INFO: (EM7180.c:64) A barometer is installed
> #000005 [ 0.013 ] INFO: (EM7180.c:66) A temperature sensor is installed
> #000006 [ 0.013 ] INFO: (EM7180.c:85) EEPROM detected on the sensor bus!
> #000007 [ 0.013 ] INFO: (EM7180.c:86) EEPROM uploaded config file!
> #000008 [ 0.013 ] INFO: (EM7180.c:88) EM7180 in initialized state!
> #000009 [ 0.013 ] INFO: (EM7180.c:91) EEPROM upload successful!
> #000010 [ 0.115 ] INFO: (EM7180.c:126) Beginning Parameter Adjustments

But why are we looking here for the MPU if we have not configured the EM7180 yet?

First we look for sensors 
`Line 457   // Check SENtral status, make sure EEPROM upload of firmware was accomplished`

We get the output that I put before, and then we start the EM7180 setup

```
Line 489
  // Set up the SENtral as sensor bus in normal operating mode
  if(!passThru) {
```

Thanks

kriswiner, 2018-08-15:
I am not sure I understand your question.

The EM7180 needs to have firmware loaded into its RAM, which is why the
first thing done is to find the EEPROM and load the firmware.

Is this your question?

kriswiner, 2018-08-15:
> Hi Kris, is it ok that my module cannot find the MPU during the SETUP()?
>
> Thanks

I don't know what you are doing so can't say.

If the EM7180 is not in pass through mode, then you will see only 0x28, the
EM7180.

In passthrough mode you should see the EM7180,  EEPROM at 0x50/0x58, the
BMP280 at 0x76, and the MPU9250 at 0x68.

If you see the BMP280 but not the MPU9250 then there is something wrong.

EinSoldiatGott, 2018-08-15:
> If you see the BMP280 but not the MPU9250 then there is something wrong.
Yes, this is what's happening.

I'll test with a second USFB Board 

Thanks

EinSoldiatGott, 2018-08-15:
And what I'm doing is porting your code to be used in other uC a CC260xx

But until now I've only ported the Setup() function. This function has a line which scans for features:
        uint8_t featureflag = readByte(EM7180_ADDRESS, EM7180_FeatureFlags);

And this is returning only barometer and temperature sensors.

> [ 0.011 ] INFO: (project_zero.c:574) Initialize i2c
> [ 0.013 ] INFO: (EM7180.c:50) ROM Ver: 58889 Should be 58889 OK
> [ 0.013 ] INFO: (EM7180.c:64) A barometer is installed
> [ 0.013 ] INFO: [...]

But until now I haven't load any data from the EEPROM maybe I haven't reached this function, maybe this is the reason, I'm in the finishing the porting of your setup() function now.

Thanks

kriswiner, 2018-08-15:
The load of the firmware happens automatically, you don;t have to do
anything except get the connections tight, etc.

[ 0.013 ] INFO: (EM7180.c:91) EEPROM upload successful!

Looks like you are doing this.

I recommend that you create a simple program that scans the I2C bus first
without doing anything (you should see 0x28) and then after you place the
EM7180 in passthrough mode (then 0x28, 0x68, 0x76, 0x50, 0x58).

If you don't get this result there is something wrong wither with our code
or the boards or your connections, etc..

EinSoldiatGott, 2018-08-15:
Hi Kris, I guess my question is wrong because the flags don't show any orientation sensor, thanks!

Now I understand that this line is not looking for any orientation sensor.

Thanks!
// Check which sensors can be detected by the EM7180
  uint8_t featureflag = readByte(EM7180_ADDRESS, EM7180_FeatureFlags);
    if(featureflag & 0x01)  Serial.println("A barometer is installed");
    if(featureflag & 0x02)  Serial.println("A humidity sensor is installed");
    if(featureflag & 0x04)  Serial.println("A temperature sensor is installed");
    if(featureflag & 0x08)  Serial.println("A custom sensor is installed");
    if(featureflag & 0x10)  Serial.println("A second custom sensor is installed");
    if(featureflag & 0x20)  Serial.println("A third custom sensor is installed");

closed 2018-08-15 by EinSoldiatGott