kriswiner/EM7180_SENtral_sensor_hub #35: firmware file

carbonadam, 2018-09-06:
Hey Kris.  I know you send out the boards with pre-loaded xxx.fw file but do we get to see a copy of it?

kriswiner, 2018-09-06:
Nope...

carbonadam, 2018-09-06:
ok, so is the guide thats on the wiki what we should use?  Im just trying to figure out how Im going to able to write to the EEPROM if it erases how all the sensors work...anyhow clueless this time

kriswiner, 2018-09-06:
You shouldn't need to write to the EEPROM for normal use. You can use the
upper portion of the EEPROM for your own use but I would advise against it
unless you know what you are doing; you do not want to erase the firmware.

There are lots of example sketches on github as well as the wiki that
should let you make use of the board.

I am not sure what your questions is exactly....