kriswiner/EM7180_SENtral_sensor_hub #37: Shipping to the UK

nyxaria, 2018-09-09:
Hi, I was wondering, how long it would take to ship the USFS mpu9250 to the UK?
https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-mpu9250/
Given that the shipping is almost £12.

Thank you.

kriswiner, 2018-09-09:
Hi George,

Shipments to the UK typically take a week or so, standard delivery for
first class international shipping (what we used to call air mail) is
typically 6 to 8 business days, bu unless the package gets stuck in customs
the actual arrival times tend to be shorter to the UK and France and a few
others.

Kris

nyxaria, 2019-04-11:
Hi Kris,

I am finally getting around to integrating the sentral into my project, but for my pcb design I would require a lbr file for EAGLE. More specifically, the one for the break outboard, so only the standard 0.254 pins would need to be modelled. Do you by any chance have this, and are willing to share?

George

kriswiner, 2019-04-11:
The design is open source and you can download it from OSH Park shared
projects page linked on the Tindie product page.

nyxaria, 2019-05-23:
Hi Kris, do you also have a 3d file, just as STEP? Thanks!

kriswiner, 2019-05-23:
You'll have to as Dylan for this since I lost all of these when my laptop
went out...

nyxaria, 2019-05-23:
Can I ask you where I can contact this fellow?

kriswiner, 2019-05-23:
From the product page:

"Just added CAD models of the board courtesy of Dylan Reynolds (
dylan.r.reynolds@gmail.com). STEP files available on request"

nyxaria, 2019-05-23:
Thank you Kris, have a great day!

closed 2019-05-23 by nyxaria