kriswiner/EM7180_SENtral_sensor_hub #38: ESP32 board + Ultimate Sensor Fusion board

afnan, 2018-09-10:
Hi,
I am using your Ultimate sensor fusion board with your ESP32 board. I would be grateful if you can answer the following:
1. Even after proper calibration including an accelerometer, if you rotate sensor at one angle (roll) more than 90, the pitch increases suddenly whereas it should remain zero or close. Why? if this is a sensor limitation or design limitation, how did HTC vive hand controllers implement it as they are using MPU6050?

2. On using 750mA lipo with ESP32, sensor fusion board and sending data through BLE, I only get 20 minutes of usage time. I am trying to understand that how POLAR wrist watch get more time from a very small battery? it too is sending data through BLE.

3. Do you think deep sleep function can help to get more juice out of this battery? if yes do you think I can use the button attached to GPIO zero on your board?

kriswiner, 2018-09-10:
1) I will assume you have calibrated all three sensors, or at least the
mag. Not sure what is happening. You will need to ask the vive people about
their device.

2) Sounds like you are not using light sleep. The ESP32 uses a lot of
power, but not more than ~100 mAH so maybe you need to turn off wifi and
practice basic power management methods. if you want to use just BLE, try
the nRF52, it uses a few mAH to run.

3) the problem with deep sleep if the ESP32 forgets its state and starts at
the top of the sketch, but yes, this should save a lot pf power depending
on your duty cycle. You may use the button on the ESP dev board as a wake
up.

afnan, 2018-09-10:
in regards to 1. It only happens if I roll more than 90 then pitch changes, however, on changing pitch roll remains steady

https://www.tindie.com/products/onehorse/nrf52832-development-board/

This looks promising but the only down side is it does not have SENTRAL :( do you think it can give accuracy of ultimate sensor fusion board? 

I only need BLE no wifi etc. 

Thanks

kriswiner, 2018-09-10:
Maybe try reading euler sngles directly from the em7180?

The fusion on the nrf52 host will be about 4 degrees heading error compared
to 2 for the em7180 but the total power usage will be <10 mA.

afnan, 2018-09-10:
Yes, I am reading hardware angles. 

Well, it is a good suggestion (I will ask my client his preference angles or battery time), I would say at this point. The client intends to go into bulk order mode in the future. Do you have any plans or can put 7180 on nrf52 board?

kriswiner, 2018-09-10:
As we discussed before, I do custom work. If interested please send me an
e-mail at tleracorp@gmail.com. Typical low power solutions with EM7180 +
BLE use < 1 mA.