kriswiner/EM7180_SENtral_sensor_hub #39: calibration & warmstart for USFS LSM6DSM + LIS2MD

leedsp, 2018-09-29:
Hi Kris,
I'm very interested in your USFS LSM6DSM + LIS2MD and going to order it from Tindie.
But I can't find any calibration and warm start supporting document or sketch for this board.
I do see you have very detailed instructions of accelerometer and magnetometer calibration and warm start for USFS MPU9250, do you have the same supporting document and sketch for USFS LSM6DSM + LIS2MD?

thanks

kriswiner, 2018-09-29:
I believe the "published" sketches for the MPU9250 will also work for the
LSM6DSM_LIS2MDL with the EM7180 in normal mode. Perhaps we will add a
sketch with warm start and calibration specifically for the LSM6DSM+LIS2MDL
version of the USFS.

CaptIgmu, 2019-02-03:
I  used @simondlevy [EM7180](https://github.com/simondlevy/EM7180) examples to calibrate and save warmstart for @kriswiner [Ultimate Sensor Fusion Solution - MPU9250](https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-mpu9250/) **_and_** its better successor [Ultimate Sensor Fusion Solution - LSM6DSM + LIS2MD](https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/). The EM7180 examples use the [CrossPlatformDataBus](https://github.com/simondlevy/CrossPlatformDataBus) and **_prior_** to Arduino-ESP32 release **_1.0.1_**, I had to change **_/src/ArduinoI2C.cpp_** to this:

```

// #define NOSTOP false     // ReSTART needed for some chips
#define NOSTOP true  // ESP32 won't work properly with ReStart I2C reads prior to r1.0.1

```

Now with release 1.0.1 I don't need to change any code, it just **_works!_**

kriswiner, 2019-02-03:
Thanks Ron...