kriswiner/EM7180_SENtral_sensor_hub #40: Overlap of Accel Calibration Regs and LPF BW Regs

braincore, 2018-10-09:
Register `0x5B` is used for both setting the accelerometer LPF bandwidth and for setting the accelerometer calibration.

From `Global.h`:
`EM7180_ACC_LPF_BW = 0x5B`
`EM7180_GP36 = 0x5B`

How does the EM7180 distinguish between the two different values being written to `0x5B`? Both writes are done after setting host control to `0x00` so my only guess is that it has to do with the magnitude of the value being written.

kriswiner, 2018-10-09:
These refer to the same register. The EM7180 reserves a few registers for
general use. We have chosen to use 36, 37, and 38 to set the MPU9250 sensor
parameters.

braincore, 2018-10-09:
FWIW, I'm using a different code base that's heavily inspired by the code in this repo so it's possible I'm doing something differently.

I noticed that my linear-acceleration samples had bias which is why I began investigating. I just removed the LPF BW code and now the linear-acceleration samples are much cleaner. So at least in my case, I suspect that the LPF writes were being interpreted as calibration parameters.

kriswiner, 2018-10-09:
I can;t say anything about your modifictions of the code but what it looks
like is that the general registers are first used to store the accel
calibration data, and then retreive the data vi warm start load, then the
general regiters are used to store sensor configuration data, all
consistent with the idea that the general registers are for transferring
data between host and the EM7180. I agree it is complicated and not as
transparent as I'd like, but it does work.

braincore, 2018-10-10:
I should have brought this up sooner, but I'm actually using the LSM6DSM + LIS2MD USFS. Any chance the GP registers used for accel calibration or LPF BW are different from the MPU9250 USFS? It was only after I read https://github.com/kriswiner/EM7180_SENtral_sensor_hub/issues/39 that it occurred to me that there might be a difference.

kriswiner, 2018-10-10:
The EM7180 registers are likely the same but the LPF and BW and sample
rates, etc are certainly not.

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/tree/master/EM7180_LSM6DSM_LIS2MDL_LPS22HB_Butterfly

braincore, 2018-10-11:
Makes sense. Thanks!

closed 2018-10-11 by braincore