kriswiner/EM7180_SENtral_sensor_hub #41: stop at "Beginning Parameter Adjustments"

robotdreamers, 2018-10-16:
I just got this new board and ran through the default program. It seems to be stuck at: "Beginning Parameter Adjustments".
What did i do wrong?

robotdreamers, 2018-10-16:
It stops here

```
while(!(param_xfer==0x4A)) {
    param_xfer = readByte(EM7180_ADDRESS, EM7180_ParamAcknowledge);
  }
```

kriswiner, 2018-10-16:
Which board? Which program? MCU? Connections?

A bit more infor would help me understand what is happening.

robotdreamers, 2018-10-16:
Sorry, this board Ultimate Sensor Fusion Solution - MPU9250. I bought from Pesky Products.

kriswiner, 2018-10-16:
Good start. How are you connecting it to your MCU, which MCU? Can you
detect the I2C address of the EM7180 on the I2C bus?

robotdreamers, 2018-10-16:
I am using Teensy 3.5. Got this message:
`Scanning...
Unknow error at address 0x01
Unknow error at address 0x02
Unknow error at address 0x03
Unknow error at address 0x04
Unknow error at address 0x05
Unknow error at address 0x06
Unknow error at address 0x07
Unknow error at address 0x08
Unknow error at address 0x09
Unknow error at address 0x0A
Unknow error at address 0x0B
Unknow error at address 0x0C
Unknow error at address 0x0D
Unknow error at address 0x0E
Unknow error at address 0x0F
Unknow error at address 0x10
Unknow error at address 0x11
Unknow error at address 0x12
Unknow error at address 0x13
Unknow error at address 0x14
Unknow error at address 0x1EEPROM detected on the sensor bus!
EEPROM uploaded config file!
EEPROM CRC incorrect!
EM7180 in initialized state!
No EEPROM detected!
Beginning Parameter Adjustments
`

robotdreamers, 2018-10-16:
I running again and I got this

`Scanning...
Unknow error at address 0x01
Unknow error at address 0x02
Unknow error at address 0x03
Unknow error at address 0x04
Unknow error at address 0x05
Unknow error at address 0x06
Unknow error at address 0x07
Unknow error at address 0x08
Unknow error at address 0x09
Unknow error at address 0x0A
Unknow error at address 0x0B
Unknow error at address 0x0C
Unknow error at address 0x0D
Unknow error at address 0x0E
Unknow error at address 0x0F
Unknow error at address 0x10
Unknow error at address 0x11
Unknow error at address 0x12
Unknow error at address 0x13
Unknow error at address 0x14
Unknow error at address 0x15
Unknow error at address 0x16
Unknow error at address 0x17
Unknow error at address 0x18
Unknow error at address 0x19
Unknow error at address 0x1A
Unknow error at address 0x1B
Unknow error at address 0x1C
Unknow error at address 0x1D
Unknow error at address 0x1E
Unknow error at address 0x1F
Unknow error at address 0x20
Unknow error at address 0x21
Unknow error at address 0x22
Unknow error at address 0x23
Unknow error at address 0x24
Unknow error at address 0x25
Unknow error at address 0x26
Unknow error at address 0x27
Unknow error at address 0x28
Unknow error at address 0x29
Unknow error at address 0x2A
Unknow error at address 0x2B
Unknow error at address 0x2C
Unknow error at address 0x2D
Unknow error at address 0x2E
Unknow error at address 0x2F
Unknow error at address 0x30
Unknow error at address 0x31
Unknow error at address 0x32
Unknow error at address 0x33
Unknow error at address 0x34
Unknow error at address 0x35
Unknow error at address 0x36
Unknow error at address 0x37
Unknow error at address 0x38
Unknow error at address 0x39
Unknow error at address 0x3A
Unknow error at address 0x3B
Unknow error at address 0x3C
Unknow error at address 0x3D
Unknow error at address 0x3E
Unknow error at address 0x3F
Unknow error at address 0x40
Unknow error at address 0x41
Unknow error at address 0x42
Unknow error at address 0x43
Unknow error at address 0x44
Unknow error at address 0x45
Unknow error at address 0x46
Unknow error at address 0x47
Unknow error at address 0x48
Unknow error at address 0x49
Unknow error at address 0x4A
Unknow error at address 0x4B
Unknow error at address 0x4C
Unknow error at address 0x4D
Unknow error at address 0x4E
Unknow error at address 0x4F
Unknow error at address 0x50
Unknow error at address 0x51
Unknow error at address 0x52
Unknow error at address 0x53
Unknow error at address 0x54
Unknow error at address 0x55
Unknow error at address 0x56
Unknow error at address 0x57
Unknow error at address 0x58
Unknow error at address 0x59
Unknow error at address 0x5A
Unknow error at address 0x5B
Unknow error at address 0x5C
Unknow error at address 0x5D
Unknow error at address 0x5E
Unknow error at address 0x5F
Unknow error at address 0x60
Unknow error at address 0x61
Unknow error at address 0x62
Unknow error at address 0x63
Unknow error at address 0x64
Unknow error at address 0x65
Unknow error at address 0x66
Unknow error at address 0x67
Unknow error at address 0x68
Unknow error at address 0x69
Unknow error at address 0x6A
Unknow error at address 0x6B
Unknow error at address 0x6C
Unknow error at address 0x6D
Unknow error at address 0x6E
Unknow error at address 0x6F
Unknow error at address 0x70
Unknow error at address 0x71
Unknow error at address 0x72
Unknow error at address 0x73
Unknow error at address 0x74
Unknow error at address 0x75
Unknow error at address 0x76
Unknow error at address 0x77
Unknow error at address 0x78
Unknow error at address 0x79
Unknow error at address 0x7A
Unknow error at address 0x7B
Unknow error at address 0x7C
Unknow error at address 0x7D
Unknow error at address 0x7E
No I2C devices found

EM7180 ROM Version: 0xFFFF
Should be: 0xE609
EM7180 RAM Version: 0x255255
EM7180 ProductID: 0xFF Should be: 0x80
EM7180 RevisionID: 0xFF Should be: 0x02
A barometer is installed
A humidity sensor is installed
A temperature sensor is installed
A custom sensor is installed
A second custom sensor is installed
A third custom sensor is installed
EEPROM detected on the sensor bus!
EEPROM uploaded config file!
EEPROM CRC incorrect!
EM7180 in initialized state!
No EEPROM detected!
Beginning Parameter Adjustments
`

kriswiner, 2018-10-16:
> Black is GND, White is 3.3V, Red is data, Yellow is Clk. The other end is
> correct as I have used this for other sensors i am testing. I just dont
> know if i need more wire?
>
> [image: img_8369]
> <https://user-images.githubusercontent.com/34329345/47051111-b3944a00-d157-11e8-849e-ff5ca367dc93.jpg>

What voltage is  showing on SDA and SCL?

robotdreamers, 2018-10-16:
Oh i don't have tool for that atm. But I have use the same connection for other sensors (which are disconnected now)

robotdreamers, 2018-10-16:
Ok it seems to work now when I change config values of Wire.begin()
`Wire.begin();//(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);`
This is what was working for me before.

robotdreamers, 2018-10-16:
Kris, do you have a piece of code to do mag and accel calib for this board? Thanks.

kriswiner, 2018-10-16:
The em7180 has auto mag and gyro calibration. Just move the device around
for a minute or so in a figure eight. You can use warm start to save the
calbration for the next power on.

kriswiner, 2018-10-16:
> Oh thanks. What about accel calib?

See the wiki.

robotdreamers, 2018-10-17:
So the system had been working quite well for couple of mins. Then after I did some warm-start calib (saving the params), now for some reason, the both software and hardware measurement stop at 1 values after few change at the beginning. It will changes a little bit if i restart then stuck at one values. Do you happen to run into this issue? Thanks.

robotdreamers, 2018-10-17:
I used the warmstart cloned from here: https://github.com/kriswiner/EM7180_SENtral_sensor_hub
Let me check more on the buffer. In any case the default value in i2c_t3.h is 
`#define I2C_TX_BUFFER_LENGTH 259`

kriswiner, 2018-10-17:
This one?

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/tree/master/WarmStartandAccelCal

robotdreamers, 2018-10-17:
I used this https://github.com/kriswiner/EM7180_SENtral_sensor_hub/tree/master/WarmStart

kriswiner, 2018-10-17:
Did you use this to both save and restore the warm start parameters?

There are several settings that would need to change for your
configuration, like wire.begin() etc. I assume you made these changes and
it was running.

Not sure what is wrong, hard to tell without exactly the sketch you are
running. Something has gone wrong. Will it run if you don;t select load
warm start?

robotdreamers, 2018-10-17:
I am trying `WarmStartandAccelCal`. And there seems to be a typo on:
this `#include "Globals.h"` instead `#include "Global.h"`

robotdreamers, 2018-10-17:
ah, yes it did run without warmstart, i.e. mode 0.

robotdreamers, 2018-10-17:
I did only `save`. I did not know there is `restore` mode as well.

robotdreamers, 2018-10-17:
Can you help tell me the x/y direction of the MPU9250 on the board? Thanks.

kriswiner, 2018-10-17:
According to the product spec, right of board is Ax, top is Ay.

robotdreamers, 2018-10-17:
After I calibrate both mag and accel following the instruction wiki.
Both are saved.
It seems like the mag calib is not working properly. When I run `EM7180_MPU9250_BMP280.ino` I got this message:
```
M7180 run status = normal mode
 EM7180 magnetic anomaly detected
 EM7180 new quaternion result
 EM7180 new mag result
 EM7180 new accel result
 EM7180 new gyro result
 EM7180 sensor status = 0
```
The the yaw is off. Do you happen to know what is going on? Thanks.

robotdreamers, 2018-10-18:
Oh I do see this too:
```
EM7180 RAM Version: 1xEM7180 ROM Version: 0xE69
Should be: 0xE609
```

kriswiner, 2018-10-18:
Tha RAM Version is correct.

kriswiner, 2018-10-18:
These messages all seem fine.

What do you mean, "the yaw is off"?

robotdreamers, 2018-10-18:
I got this message `EM7180 magnetic anomaly detected`
while I am expecting `EM7180 mag calibration completed` 
The yaw heading is not the same as the one i got when I just done the calibration, off for 100 degree or something.

kriswiner, 2018-10-18:
Did you check your mag calibration like in this
<https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration>
writeup?

How do you know your sensors are calibrated? How did you check?

kriswiner, 2018-10-18:
When properly calibrated the device is an absolute orientation estimator.
Proper yaw comparison is to point the top of the boards toward True North
and check the yaw, it should read zero if you have properly compensated for
your local magnetic declination.

robotdreamers, 2018-10-18:
I got this very weird thing happened. After following the instruction carefully, I can get the heading error within 2 degree. Please see the picture of the screenshot and my setting. However, after I saved the config and after few mins in that instead state, the heading yaw now becomes 18 degree. I try to recalibrate mag again but it is still 18 degree after calibration. Not sure what went wrong.

[image]
[image]

kriswiner, 2018-10-18:
> ![img_3140](

Not sure, is there a magnet nearby or something that could affect the
heading?

Are you sure the mag is properly calibrated when you store the warm start
parameters?

One way to tell is look at Mz when the board is facing upo and compre to Mz
when the board is facing down, they should be equal but opposite in sign.
If not then the mag is not calibrated.

robotdreamers, 2018-10-18:
Does the chip has some self-calibration capability? After saved the mag calib, I let it running next to my small compass shown in the previous picture. Will that created the small drift overtime?

kriswiner, 2018-10-18:
> and indeed, the Mz in my case is not complete opposite in sign 210 vs -250.

So your calibrtion is off by ~20%.

Yes, the EM7180 does auto gyro and mag calibration as I told you.  if the
mag is not well calibrated then the EM7180 will continue to  try to
calibrate even after reading warm start parameters so this might be why you
are seeing changes.

robotdreamers, 2018-10-18:
is there an option to stop the self-calibration? assuming I have an application where this is necessary.

robotdreamers, 2018-10-18:
Self-calib would work if the chip is moving/changing direction. But if it
stays still, it won't be that helpful. Am I right?

kriswiner, 2018-10-18:
No way to turn this off.

robotdreamers, 2018-10-18:
after understanding how it works especially the self-calib capability. I think i finally got it working stably.  Thank Kris. I just ordered two more USFS boards.

kriswiner, 2018-10-18:
I am sorry you had so much trouble. Maybe you could summarize your findings
for others who have similar trouble in the future? Maybe as a review of the
board on Tindie?

kriswiner, 2018-10-21:
> Yes, I will write some review note when I am done.
>
> One more question, it seems like your code do not calibrate the gyro. I
> did not find the part of the code that is similar to function
> MPU9250::calibrateGyro in the standard Adruino MPU9250 library.

Gyro cal is automatic