kriswiner/EM7180_SENtral_sensor_hub #42: Using quat output from Sentral chip

robotdreamers, 2018-10-26:
Hi Kris, I want to do find the rotation from quat1 to quat2. I used some standard lib like pyquaternion but got wrong result. When I use your function to convert from quat to rpy and rads, everthing is good. So it seems like the quat output from Sentral may have something special that makes it not working with pyquaternion. I wonder if you can give us some pointer to solve this problem. More specifically, we need to operation: quat multiplication and inverse. Thank much.

kriswiner, 2018-10-26:
Not sure what the trouble is, except that maybe you need to make sure which
is Qw and which is qxyz? Maybe these are reversed or something.

robotdreamers, 2018-10-26:
True, the output of Sentral is qxyz, qw, which seems to be reversed of standard normal.
BTW, can you share with us the reference to your formula to calculate rpy from quat? Thanks.

robotdreamers, 2019-02-24:
Hi Kris, do you have discount if I buy 10 items?
 We want to created a glove like this one.
We want to do it by ourselves first. Appreciate any advice. Thank you.

kriswiner, 2019-02-24:
I can offer 10% discount, so buy 9 get 1 free if you pay direct via paypal
to tleracorp@gmail.com. Please include US shipping $3.50 or Internaltional
shipping $14.95. Please send your shipping address including phone number.

The EM7180 breakouts have only one I2C address so it will require an I2C
multiplexer to get data from 10 of them on a single I2C bus.

Also, this is new and might help:
https://github.com/gregtomasch/EM7180_SENtral_Calibration

robotdreamers, 2019-02-24:
You can ship to Vietnam, right?
~Bach

robotdreamers, 2019-02-24:
Can you give me total price to ship 10 to VN?
I will give you the address in the PayPal transaction.
Thanks.

~Bach

kriswiner, 2019-02-24:
I have done before using regular first class air mail.

robotdreamers, 2019-02-25:
Kris, I sent the money.
The address for the package is:

*Tran Bich Khanh*
*P907, Toa nha 3B Phuong Liet, Thanh Xuan*
*Ha Noi, Vietnam*
*Tel: +84-91-2353334*

Thanks.
~Bach

kriswiner, 2019-02-25:
Is there a zip code or province or city code number?

kriswiner, 2019-02-25:
Customs number UA428231946US

kriswiner, 2019-02-25:
Also, which variant do you want? MPU9250 or LSM6DSM+LIS2MDL?

robotdreamers, 2019-02-25:
Kris, sorry my team had different opinion and don't want to order this for
now. They found a cheaper solution.
Could you please send us the money back? Please take out any fee that you
find appropriate.
Very sorry for the inconveniences.