kriswiner/EM7180_SENtral_sensor_hub #43: NRF52 code

afnan, 2018-11-01:
Hi,

I am planing to switch from MPU9250 based device to EM7180_LSM6DSM_LIS2MDL_LPS22HB. This time I have a constraint that I must use nrf SDK as iwill be working on nrf52 device. Do you have any example working code for that?

Thanks

kriswiner, 2018-11-01:
No, but we have an Arduino sketch that can run on the nRF52. Or you can
translate it to work with the nRF52 SDK.

afnan, 2018-11-01:
If it is this https://github.com/kriswiner/EM7180_SENtral_sensor_hub/blob/master/EM7180_LSM6DSM_LIS2MDL_LPS22HB_Butterfly/EM7180_LSM6DSM_LIS2MDL_LPS22HB_Butterfly.ino

I have the following questions.
1. Why do you have references to "USFS.h" in the file? this file (as per comments) contains information of MPU9250 stuff?
2. Why are you using RTC ?

kriswiner, 2018-11-01:
Yeah, we really need to update this. This was a mashup of ST sensor files
with MPU9250-based files. It works fine for the EM7180 in normal mode but
we need to create a sketch that is specific for the LSM6DSM_LIS2MDL.

We use th RTC alarm at 1 Hz to send data to the serial monitor.

It should be straightforward to use this as a template to create an Arduino
sketch for the nRF52.