kriswiner/EM7180_SENtral_sensor_hub #44: How are you gettting the hardware quaternions?

XaviLC, 2018-11-15:
Hello @kriswiner ,

I am newbie on the rotation topic using quaternions, I have been reading abot then and the different ways of representing rotation changes, so sorry in advance if my questions might be quite simple.

I have been looking your [code ](https://github.com/kriswiner/EM7180_SENtral_sensor_hub/blob/master/EM7180_BMI160_AK8963C.ino) for the BMI160, device that I am using, and there is a point where obtaining the quaternions that I get a bit lost.

I see that you have defined to set of quaternions, hardware quaternions `Quat[4]` and software quaternions `q[4]`. Could you explain me how you are obtaining the hardware quaternions? I am especially interested in q0 (`Quat[0]`), is this value given only by EM7180 sensor or it can be retreived as well from BMI160?

And finally the  software quaternions `q[4]` are coming from the application of the `MadgwickQuaternionUpdate()` method?

Thanks in advance!

kriswiner, 2018-11-15:
Yes, hardware qs from the EM7180 and software qs from the madgwick filter.

XaviLC, 2018-11-16:
Thanks for replying,

Then that means that BMI160 doesn't provide the value q0 from hardware and if I use this sensor I would have to make use of just the software base right? 
One more question regarding to q0, I have seen other repo of yours for the MPU9250, this sensor also provides the q0 as far as I understood from your code, but I could not find any explicit register in the register map of the MPU9250 that holds q0. Could you kindly point me out where or what I should look to find that register?

Thank you very much,

REgards!

kriswiner, 2018-11-16:
Tne BNO055 is what you want if you want to get 9 DoF quaternions from one
device. But it is not as accurate as some other solutions.

The MPU9250 can provide 6 DoF quaternions but doesn't use the magnetometer
data in the orientation solution. Using it's DMP is not easy.

This
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-mpu9250/>
the most accurate quaternion engine you can buy at this size and price
range.