kriswiner/EM7180_SENtral_sensor_hub #45: magnetic anomaly detected

Rababermaber, 2018-11-22:
Hello,

whatever I am doing I always get the report that there is a magnetic anomaly detected.
I am using a Teensy 3.6 and added the "i2_t3" library and optimized the USFS library to work with it.

kriswiner, 2018-11-22:
Ok, and this is a problem, how?

Rababermaber, 2018-11-22:
[Issue #29](https://github.com/kriswiner/EM7180_SENtral_sensor_hub/issues/29#issuecomment-411199798)
I thought so because I read in the Issue linked above that if "magnetic anomaly detected" the magnometer is ignored.

kriswiner, 2018-11-23:
Shouldn't be an issue if you calibrate the sensors properly and use warm
start... Are you not getting accurate orientation estimation?

Rababermaber, 2018-11-23:
currently without warm start kind of good - I just got the sensor a few hours ago so I am going to fully test it tomorrow

Rababermaber, 2018-11-23:
The shaking and the magnetic anomaly detected problems are still present, also when testing today the YAW value slowly increased over time and by correcting to the "correct" heading my robot just performed circles...

kriswiner, 2018-11-23:
Not sure what the shaking problem is exactly. How did you calibrate the
sensors? Some more details could be useful to determine what kind of help
you might need.

Rababermaber, 2018-11-25:
For the magnometer calibration I'm waving the device in an figure 8, but the AlgorithmStatusRegister is never showing me that the magnometer calibraton is completed, instead all the time it shows me that there is a "magnetic anomaly", like I described it in the beginning.

My shaking problem is, that if I shake the sensor, like turning it 90° rapidly (and then bringing it back to its original position), the YAW value jumped to another position (usually about +-20°) and then stays there.

kriswiner, 2018-11-25:
This all sounds like you are not calibrating the megnetometer (and possible
the accelerometer) sufficiently. Can you simply check the calibration
yourself? Maybe this
<https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration>
would help: it describes a simply way to check the calibration quality
without relying on a calibration bit.

Rababermaber, 2018-11-26:
I am trying to calibrate the sensor without any success! I am using your code (WarmStartandAccelCal) and followed the steps described in the wiki section F, but no matter how long I tried to calibrate the magnetometer it never showed me any good results. Also the software and hardware yaw values nearly always differed for more than 10°/20°/... . When restarting the board after saving some "calibration data" the yaw value never stayed constant when heading the sensor in the same direction that it faced before the restart. I have absolutely no idea what i could have done wrong D:

Thank you for your help

kriswiner, 2018-11-26:
It would help to know which MCU you are using. And if you would send a copy
of your sketch (zip file, don;t cut and paste here. if you need to you can
send an e-mail to tleracorp@gmail.com.

kriswiner, 2018-11-27:
> EM71280_MPU9250_BMP280_M24512DFC_WS_Acc_Cal.zip <http://Sketch>
>
> There is not much code I could send you, I am mostly using the code
> provided in this Git. The code I linked above is exactly the one I used
> when trying to calibrate the sensor (I have not changed much other than the
> INT pin and some I²C stuff). I am using a Teensy 3.6 and I am using this
> sensor I bought from you #link
> <https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/#specs>
> ;

OK, but this sensor does not use the MPU9250, so why are you using a sketch
that supports the MPU9250?

Rababermaber, 2018-11-27:
Ok I see, but that explains why the WarmStart did not work. But for all my other testing i was using the code provided for the sensor baord. Is there an WarmStart sketch I could use for my sensor?

kriswiner, 2018-11-27:
I would start with getting the sensors properly calibrated and getting
sensible heading behavior out of the EM7180 before I would worry about warm
start.

Can you calibrate the sensors? Can you confirm the sensors are calibrated?
When the sensors are properly calibrated, do you see reasonable orientation
estimation? If not, specifically what is the problem (a plot of yaw vs
actual heading or time would help).

It's pretty hard to know from the little you have told me what the
problem(s) might be.