kriswiner/EM7180_SENtral_sensor_hub #47: CpuReset

Julspower, 2019-01-19:
Im trying to use the SENtral using rpi zero w
connected correctly I can read version and everything
but im I try to use it I get EM7180_EventStatus == 0x01

at start I do
    wiringPiI2CWriteReg8(ni2cFD, EM7180_ResetRequest, 0x01);
    while(wiringPiI2CReadReg8(ni2cFD, EM7180_ResetRequest) != 0x00)
        QThread::msleep(250);
    while(1)
        n = wiringPiI2CReadReg8(ni2cFD, EM7180_EventStatus); // reading clears the register
I always read n at 0x01 when in the debugger any idea?

kriswiner, 2019-01-19:
Nope, never use Raspberry pi.

Does the SENtral produce data as expected?

What does the RAM version say? is it reading the EEPROM?

Julspower, 2019-01-19:
The event status register never report quaterions data 
I did not try to read the quat register (even if no new data is reported)
Ram version 1 and 2 are: 0xAE, 0x23

kriswiner, 2019-01-19:
RAM version should be 0x17435  if the EEPROM is being properly read as here
(scroll down):

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/C.-Managing-the-Configuration-File

Do you see the EM7180 on the I2C bus at 0x28?

Is it in normal (master) mode?

It should read the EEPROM automatically in master mode and the code should
wurey the ROM, RAM versions as well as the products and revision IDs to be
sure of proper communication between HOST and EM7180 and between EM7180 and
EEPROM.

Julspower, 2019-01-20:
Yes I see it at 0x28
the status bit tell me the eeprom has been loaded: EM7180_SentralStatus 0x01 != 0
I do set EM7180_HostControl to 0x01 and read it back to 0x01, but after the bit cpureset has been set it return to 0x00

`    cRam1 = wiringPiI2CReadReg8(ni2cFD, EM7180_RAMVersion1);
    cRam2 = wiringPiI2CReadReg8(ni2cFD, EM7180_RAMVersion2);

    //Attend l'EEPROM
    n = wiringPiI2CReadReg8(ni2cFD, EM7180_SentralStatus);
    while((n & 0x01) == 0)
    {
        QThread::msleep(250);
        n = wiringPiI2CReadReg8(ni2cFD, EM7180_SentralStatus);
    }
    cRam1 = wiringPiI2CReadReg8(ni2cFD, EM7180_RAMVersion1);
    cRam2 = wiringPiI2CReadReg8(ni2cFD, EM7180_RAMVersion2);
`
here the first time I read cRam1 and 2 they are at 0
the second time I read 0xAE23

ProductID = 0x80
RevisionID = 0x02
Could it be a wrong eeprom config?

my I2C clock between host and SENtral is 400khz

Julspower, 2019-01-26:
Found out that
[https://github.com/simondlevy/EM7180](This) works
Must be my initialization that caused problem

closed 2019-01-26 by Julspower