kriswiner/EM7180_SENtral_sensor_hub #49: ESP32, I2C ok, Beginning Parameter Adjustments

jadamc, 2019-02-02:
1. ESP32 Connected to Sentral MPU9250 + BMP280 from tindie
2. I2C 0x28 found okay
3. Code using: /* EM7180_MPU9250_BMP280_t3 Basic Example Code
4. Code adjusted on the Wire commands
5. Connections: +3v3, GND, SDA, SCL, INT
6. I2C also connected to OLED display on I2C 3C which works fine.

Every time the below is the output on the Serial.  It boots up, cannot find any data in reading the ROM, writes the EEPROM (apparently successfully), the start Begining Parameter Adjustments and nothing more, getting trapped in the code at:

while(!(param_xfer==0x4A)) {
    param_xfer = readByte(EM7180_ADDRESS, EM7180_ParamAcknowledge);

Any help would be very much appreciated.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
Scanning...
I2C device found at address 0x28  !
I2C device found at address 0x3C  !
done

EM7180 ROM Version: 0x00
Should be: 0xE609
EM7180 RAM Version: 0x00
EM7180 ProductID: 0x0 Should be: 0x80
EM7180 RevisionID: 0x0 Should be: 0x02
EEPROM upload successful!
Beginning Parameter Adjustments

kriswiner, 2019-02-02:
Looks like you are not reading the EEPROM.

CaptIgmu, 2019-02-03:
The usual culprit with these type of errors on the ESP32 is **_I2C._** There has been a long troubled history of Arduino-ESP32 I2C problems until the latest **_release 1.0.1_**. With this release, I2C now is much more stable, thanks to the efforts of @stickbreaker. One main problem before was ReStart I2C **_reads_** and could be fixed by changing this:

```

// #define I2C_NOSTOP false  // For ReSTART to keep connection Alive
#define I2C_NOSTOP true     // ESP32 needs this to work! No ReSTART

. . .

Wire.endTransmission(I2C_NOSTOP);  // Send the Tx buffer, but send a restart to keep connection alive

```

I also used @simondlevy [EM7180](https://github.com/simondlevy/EM7180) examples to calibrate and save warmstart for @kriswiner [Ultimate Sensor Fusion Solution - MPU9250](https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-mpu9250/) **_and_** its better successor [Ultimate Sensor Fusion Solution - LSM6DSM + LIS2MD](https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/). The EM7180 examples use the [CrossPlatformDataBus](https://github.com/simondlevy/CrossPlatformDataBus) and **_prior_** to Arduino-ESP32 release **_1.0.1_**, I had to change **_/src/ArduinoI2C.cpp_** to this:

```

// #define NOSTOP false     // ReSTART needed for some chips
#define NOSTOP true  // ESP32 won't work properly with ReStart I2C reads prior to r1.0.1

```

Now with release 1.0.1 I don't need to change any code, it just **_works!_**

jadamc, 2019-02-04:
That fixed it.  Thank you.

efremovru, 2021-02-15:
Kris, i have very similar problem.  Tried the suggestion above, no change.

I am trying to use USFS with ESP32. I am using VSC & Platformio. I (think) that I made the necessary adjustments and flashed the code to the chip. This is the result

EM7180 ROM Version: 0xE69
Should be: 0xE609
EM7180 RAM Version: 0x17435
EM7180 ProductID: 0x80 Should be: 0x80
EM7180 RevisionID: 0x2 Should be: 0x02
A barometer is installed
A temperature sensor is installed
EEPROM detected on the sensor bus!
EEPROM uploaded config file!
EM7180 in initialized state!
EEPROM upload successful!
Beginning Parameter Adjustments

And that’s all she wrote. I added a serial print to see the value of 

param_xfer = readByte(EM7180_ADDRESS, EM7180_ParamAcknowledge);

and it is 0, which is a problem because it is supposed to be 4B

Any idea what am I doing wrong?

this is the code I am using
https://github.com/kriswiner/EM7180_SENtral_sensor_hub/blob/master/EM7180_MPU9250_BMP280.ino

Thanks in advance!

kriswiner, 2021-02-15:
"  flashed the code to the chip"

What does this mean? Loaded firmware to the EM7180?

Sounds like you have the wrong firmware for your sensors suite...

efremovru, 2021-02-15:
Apologies, i think that i misused the word. What i meant is load the code on the chip. I used the same process i always use for this chip. It worked perfectly well with a "generic" MPU9250 

https://smile.amazon.com/HiLetgo-Gyroscope-Acceleration-Accelerator-Magnetometer/dp/B01I1J0Z7Y/ref=sr_1_3?crid=1UWCX0X1E17U6&dchild=1&keywords=mpu9250&qid=1613404533&sprefix=MPU9250%2Caps%2C256&sr=8-3

and one of your other sketches

https://github.com/kriswiner/MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

it worked without a glitch but the accuracy was not there. that's why i thought that using USFS will solve my problem

kriswiner, 2021-02-15:
I am still a bit confused here. Did you buy one of the USFS from my Tindie
store and then try to make it work using an Arduino sketch?

If so, the best source of properly working sketches for the USFS can be
found here <https://github.com/gregtomasch/EM7180_SENtral_Calibration>.
Please try the one appropriate for your MCU and if you are still having
trouble, please e-mail me at tleracorp@gmail.com and we will sort it out.

It is possible (and likely) that some of the older EM7180 sketches are no
longer compatible with the latest firmware.

efremovru, 2021-02-15:
Kris, thanks for the quick reply. Yes, i got the USFS from your tindie store about a month ago, I downloaded the Calibration sketch you are referencing but i thought that i don't need to use it. Will give it a try and report back.