kriswiner/EM7180_SENtral_sensor_hub #50: Raspberry Pi calibration options

JNatael, 2019-03-06:
I'm currently trying to implement the device on a Raspberry Pi based on this library. I've managed to implement data collection just fine, but I'm a bit unclear about calibration options. The documentation seems to only have sketches for a Teensy device, and the pi code itself has a non-Master object type that seems to write to individual registers and such but if that is usable for calibration, and how to use it, I'm unable to find the documentation or figure it out.

Is calibration with the Pi possible? If so, how can I go about learning how to do it? If not, when will it be added?

kriswiner, 2019-03-06:
No plans to specifically support the Raspberry pi, we don't use it. But if
you can read data from the EM7180 with the pi there is no reason you can't
do everything else with it using the pi including calibrate it. Please see
this <https://github.com/gregtomasch/EM7180_SENtral_Calibration> and this
<https://github.com/simondlevy/EM7180>. If still not good enough, we might
be available to make this work for you on a contract basis.

mhmhartman, 2019-08-21:
@JNatael have you made any progress on this?