kriswiner/EM7180_SENtral_sensor_hub #51: Pitch and Roll semstivity

afnan, 2019-03-27:
Hi,

I enjoy working with your modules and they are very much accurate. This is the Testament of the time you spend on designing these things. 

I am trying to understand why following peculiar behavior occurs with the new IMU boards. 

When the pitch is rotated to near 90 degrees, roll becomes very very sensitive and shoots to more than 100degrees. 
When the roll is rotated either to + or - 90 degrees, pitch remains near zero - which is expected. 

If it is the design limitation of MEMS based sensor, then it should happen for both but it does not. I am not using any code. They came pre-programmed. (ESP32 + IMU)

kriswiner, 2019-03-27:
"They came pre-programmed. (ESP32 + IMU)"

Yes, but they might not be equally-well calibrated, or calibrated at all.

How did you calibrate them?

Also, when the Euler angles reach their extreme, the values aren't reliable.

This is the same problem as determining the direction of South when
standing on the North pole.

afnan, 2019-03-28:
I have a warmstart sketch that i use. "This is the same problem as determining the direction of South when
standing on the North pole." this makes sense