kriswiner/EM7180_SENtral_sensor_hub #53: Best practices for tracking drift and measuring performance

carbotaniuman, 2019-04-22:
Hello Kris,

I'm currently developing an IMU for a closed ecosystem and I wanted to compare my solution to existing COTS products (mainly heading wise). Do you have any tips on accomplishing this?

Thanks

kriswiner, 2019-04-22:
Get a low-cost rotary stage for reproducible testing, like this
<https://hackaday.com/wp-content/uploads/2019/03/hackaday_journal-gregorytomasch_kriswiner-heading_accuracy_using_mems_sensors.pdf>
one. Put your solution together with other COTS solutions on the same pcb
for ease of testing. Develop a standard calibration procedure, etc.

carbotaniuman, 2019-05-21:
@kriswiner I got something set up. Is there anything better for 6DoF accel + gyro fusion than an Extended Kalman Filter?

kriswiner, 2019-05-21:
Better, don't know. Different, there is this
<https://github.com/kriswiner/MPU6050/blob/master/quaternionFilter.ino>.