kriswiner/EM7180_SENtral_sensor_hub #54: Duplicate Calibration

afnan, 2019-04-25:
Hi,
I want your opinion on minimizing the calibration time for production. If I calibrate one device accurately, do you think it is a good idea to replicate the calibration from one device to others, i.e., copying calibration programmatically?

kriswiner, 2019-04-25:
This can't possibly work. The calibration is device specific.

afnan, 2019-05-01:
I see. So all I have to do in order to change the calibration from MPU9250 to the new boards, I need to change the register addresses? 
https://github.com/kriswiner/EM7180_SENtral_sensor_hub/tree/master/WarmStartandAccelCal

in above program? or do we have any WS anc Acc Calibration program written for the new boards?

kriswiner, 2019-05-01:
Not sure what new boards you mean, but see this
<https://github.com/gregtomasch/EM7180_SENtral_Calibration>.

afnan, 2019-05-02:
> Not sure what new boards you mean, but see this <https://github.com/gregtomasch/EM7180_SENtral_Calibration>.
> […](#)

i meant for these _LSM6DSM_LIS2MDL_LPS22HB. do i use same program or i have to update somethinh

afnan, 2019-05-02:
I found on that url. THanks

closed 2019-05-02 by afnan