kriswiner/EM7180_SENtral_sensor_hub #56: Roll and Pitch Accuracy levels

sls12, 2019-05-05:
Thanks for things up here. 

Curious about the kind of roll and pitch accuracy levels that can be obtained with MPU-9250/ICM 20948 or any of the ST MEMS. 

Generally it is believed to be <1 deg, though haven't found anything specific. Is it possible to get better than that?

Thanks.

kriswiner, 2019-05-05:
In all of our testing of heading we generally see pitch and roll at 1
degree error or less. We have recently procured a 2-axis goniometer and are
doing some testing of heading as a function of different pitch and roll
angles. But generally we would not test pitch and roll accuracy
specifically.

If your goal is to estimate pitch and roll only, then using an accel/gyro
like the LSM6DSM with proper calibration is enough to rather easily achieve
the <1 degree rms standard. How much better you could do? I do not know.
What is the requirement?

sls12, 2019-05-06:
Looking at something close to 0.1 degrees for roll, pitch and yaw. May be its a big ask!

kriswiner, 2019-05-06:
The real questions is, how well can you do with a good sensor suite and
proper calibration methods? In other words, the answer you need will only
come from actual testing.

sls12, 2019-05-06:
Yea. That makes sense. What would be your suggestion for a good sensor suite?

kriswiner, 2019-05-06:
LSM6DSM

closed 2019-05-06 by sls12