kriswiner/EM7180_SENtral_sensor_hub #57: Suddenly missing EEPROM

Nicrodk, 2019-05-10:
Hello Kris, I have been using the USF for a few weeks now for my semester project and it was working perfectly but then suddenly it stopped being able to load up the EEPROM and when I loaded up the EM71280_MPU9250_BMP280_M24512DFC_WS_Acc_Cal.ino this is what I ended up with 
[image]
Our project has the USF inside of a drumstick and we used some cotton wool to make sure that it wouldn't be damaged inside and as mentioned it has been working for a few weeks but now suddenly it seems like it cannot find the EEPROM, any suggestions on how to approach this?

kriswiner, 2019-05-10:
Looks like the EEPROM has been damaged. I suggest you replace the USFS
board to get g oing again and inspect the damaged board to find out what
caused the damage to avoid it in the future.

Nicrodk, 2019-05-13:
Alright, thanks for the response, we'll get to it and see if we can identify what would've damaged the component

closed 2019-05-13 by Nicrodk