kriswiner/EM7180_SENtral_sensor_hub #58: Barometer max out at 10K feet (-32768 raw)

AllDigital33, 2019-08-18:
I am using the Ultimate Sensor Fusion to track telemetry on hobby rockets. I did two launches this weekend over 10K feet and noticed the barometer data crapped out over 10K feet ASL. I thought that was odd, since I did two previous launches using a BMP180 alone (without the EM7180 board) with no issue. When I reviewed the code and configuration today, I realized the raw pressure data is an Int16, so it will max out at -32768, which is around 685 mbar. I looked through all the documentation, but could not find anything that allows setting a range or anything that limits the actual range of the BMP180. The sensor should be accurate to 300 mbar, so I assume there is a setting somewhere that is being overlooked. Please advise.

AllDigital33, 2019-08-18:
Correction:  I meant BMP280 above and not BMP180.

kriswiner, 2019-08-18:
I think the problem is that when we set up the EM7180 to manage the
barometer we did not consider we would ever need more than int16_t data, so
this sets the limit of what is possible in this (master) mode. If you want
to use the BMP280 at its full resolution you will have to make use of the
pass through mode to read the barometer.