kriswiner/EM7180_SENtral_sensor_hub #61: Missing writeByte(EM7180_ADDRESS, EM7180_AlgorithmControl, 0x00); ?

gerowtj, 2019-09-21:
In the "EM71280_MPU9250_BMP280_M24512DFC_WS_Acc_Cal.ino" program, routine "void EM7180_set_WS_params()" there is not a "writeByte(EM7180_ADDRESS, EM7180_AlgorithmControl, 0x00);" at the end of the routine to restart the algorithm.  

Should there be a write of 0x00 to the AlgorithmControl register after the "writeByte(EM7180_ADDRESS, EM7180_ParamRequest, 0x00);" at line 929?

Thanks,
Tom

kriswiner, 2019-09-21:
Way old sketches, try these
<https://github.com/gregtomasch/EM7180_SENtral_Calibration>.

kriswiner, 2019-09-21:
In general, WS_Resume is called after WS_Passthru and there is no need to
restart the fusion algorithm while the EM7180 is in passthrough mode.

gerowtj, 2019-09-21:
Perfect makes sense, thanks for the link to the other files, and the quick reply!

Closing this issue

Regards,
Tom

closed 2019-09-21 by gerowtj

K-Space909, 2021-03-17:
So is this out of date then?

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/tree/master/WarmStartandAccelCal

I've been using this sketch since I began with the USFS board, since it was linked in the writeup found here:

kriswiner, 2021-03-17:
Probably. Try this
<https://github.com/gregtomasch/EM7180_SENtral_Calibration> instead.

K-Space909, 2021-03-19:
I can't seem to get this one to work. Using Teensy 3.2, the Sentral MPU9250 board, and the "Teensy_INV_USFS_Baseline_Calibration_Utility_Acc_WS" script with all the header files copied into my VisualMicro solution in Visual Studio. 

I was able to get "Sentral Status: 11" at first, and tried to calibrate the accel, but after rebooting I still had "Acc Cal data invalid". And now I'm only getting "Sentral Status: 3".
```

Opening port
Port open
VMDPR_Initializing Sentral_0
Sentral_0:
Scanning...
I2C device found at address 0x28  !
I2C scan complete

Sentral Status: 3 (Should be 11)

Sentral firmware loaded. Fetching Accel Cal and WS parameters...
Done...

Acceleromater Calibration Data:
X-acc max: -1
Y-acc max: -1
Z-acc max: -1
X-acc min: -1
Y-acc min: -1
Z-acc min: -1

Checking/loading Acc Cal data...
Acc Cal data invalid! Defaults loaded...
Done. Loading Warm Start Parameters, modifying sensor ranges and data rates...
Warm Start data NOT loaded!
Done. Starting the Sentral...
Done. Loading algorithm tuning parameters...
Sentral initialization complete!

Algorithm Status = 0

ax_0 = -8 ay_0 = 4 az_0 = 1036 mg
gx_0 = 0.00 gy_0 = 0.00 gz_0 = 0.00 deg/s

mx_0 = 22 my_0 = -22 mz_0 = 1 uT

Sentral_0 Quaternion (NED):
Q0_0 = 0.91 Qx_0 = -0.03 Qy_0 = -0.06 Qz_0 = 0.41

Sentral_0 Yaw, Pitch, Roll: 62.53, -5.09, -6.25

Baro Pressure: 949.40 mbar
Baro Temperature: 19.49 deg C

Loop Cycletime:11087316 us

Send '1' for Sentral_0 Accel Cal
Make sure the desired accelerometer axis is properly aligned with gravity and remains still
All three accelerometers must to be calibrated in the +/-1g condition for accurate results
Send '2' to save Sentral_0 Warm Start params
```

K-Space909, 2021-03-19:
I probably need to hook up the interrupt pin, don't I?

kriswiner, 2021-03-19:
Looks like your attempt at storing the warm start configuration didn't
"take" for some reason. Otherwise, the code seems to be running at least.

Not sure what the problem is exactly?

Are you trying to do the accel cal?

kriswiner, 2021-03-19:
If the sketch you are trying to use is expecting an interrupt then yes.

K-Space909, 2021-03-19:
Copy that. I'm still using the sketch as-written, and will refactor it just a bit as I bring it into my solution. Do you have any general recommendations for changing the script to a more request-based format? My scripting is a little slow since I'm sending packets over the air at around 80Hz, and I want to be sure I'm sending the most recent IMU data with each packet.

kriswiner, 2021-03-19:
Polling instead of interrupt works for this. Or just set up a time-based
data read in your main loop and read the data when you wnt. As long as your
reads are much slower than the data rate, you will get the latest data as
of the time of read request.