kriswiner/EM7180_SENtral_sensor_hub #66: USFS wiht ICM-20948?

roverwing, 2020-04-11:
Just a question:
have you ever tested Invensense latest 9DOF chip, ICM-20948, as the sensor for Ultimate Fusion board? One would expect that it could give better performance than 9250, but it would be interesting to verify...

Thank you for your products!

kriswiner, 2020-04-11:
0No, but we do have a new USFS
<https://www.tindie.com/products/onehorse/max32660-motion-co-processor/>
that is 10x better than out USFS-MPU950 in terms of heading accuracy.

The problem with the ICM20948 is the dual power rail requirement...messy.

roverwing, 2020-04-11:
Wow! I missed that one - the first page of your Tindie store doesn't show it (probably because it is priced higher...)
Most impressive.