kriswiner/EM7180_SENtral_sensor_hub #67: USFS MPU9250 without barometer

mpava, 2020-04-19:
Hi Kris,

Thanks for making the USFS board. It works really well, I am really impressed with the quaternion precision. I am designing a custom board which doesn't require a barometer, the core design is same as the USFS minus the barometer. I am using the config file from the USFS. All setup and parameter initialisation of EM7180 works fine up to the point when I want to read the quaternions. I get EventStatus == 2, SensorStatus == 0x80 and ErrorRegister == 0x28. I assume this is due to the missing barometer. The 0x28 error code doesn't seem to be documented in the PNI docs, so I am slightly guessing here.

Is the EM7180 firmware hardwired to look for the baro device and when it can't find it, it gets stuck? Do you know is there a way to disable byro reading in the algo or would you have a config file that doesn't need a byro? In pass-through mode, interacting with MPU9250 and the EEPROM   works perfectly well without problems. I tried setting EM7180_BaroRate (0x58 register) to zero hoping to disable it that way without luck.

Any thoughts would be much appreciated.

thanks

Mike

kriswiner, 2020-04-19:
Hi Mike,

The firmware for the commercial EM7180 (USFS) does indeed expect a
barometer and if it is not accessible on the master I2C bus will throw an
error flag and stop working. As far as I know, there is no way to disable
this behavior with the given firmware. Of course, you could add the baro,
or we could create a version of the firmware that doesn;t use the baro, but
we would have to charge you for the custom firmware.

Kris

mpava, 2020-04-19:
Hi Kris

Thanks for your quick reply. I would really prefer running the circuit without the baro. I would be interested in the custom firmware. How can I contact you offline?

thanks

Mike

kriswiner, 2020-04-19:
Send me an e-mail at tleracorp@gmail.com.

mpava, 2020-04-19:
thanks Kris. Email sent.