kriswiner/EM7180_SENtral_sensor_hub #70: latestResults

kriswiner, 2020-12-08:
[image]

gerowtj, 2020-12-14:
Hi Kris, is this the results of new firmware algorithms on the USFS board?  Or is this from additional Arduino software filtering enhancements?  

It appears you have eliminated a good portion of the systematic sinusoidal error that was shown in the linked pdf.

I'm basing my statements off the results from the [https://hackaday.com/wp-content/uploads/2019/03/hackaday_journal-gregorytomasch_kriswiner-heading_accuracy_using_mems_sensors.pdf](url) 

Regards,
Tom

kriswiner, 2020-12-14:
Hi Tom,

Not sure what 'this' refers to, but the improvement in heading acvuracy is
due to our own calibration and sensor fusion methods. We have embedded
these in a motion coprocessor using Maxim's MAX32660 that we call the
USFSMAX available now on Tindie.

kriswiner, 2020-12-14:
See [here](https://www.tindie.com/products/onehorse/usfsmax-module-and-carrier/).

gerowtj, 2020-12-14:
Sorry about the ambiguity.  The 'this' is in reference to the chart you posted.  I didn't realize there was a new product, the USFSMax available.  I should have figured that out based on the title of the graph.

Looks like I need to add another item to my Christmas list ;-)