kriswiner/EM7180_SENtral_sensor_hub #71: Best practices for portable units with 18650 batteries

gerowtj, 2020-12-10:
Hi Kris,
The USFS is working very well with an STM32 "blue pill" clone.  I am very impressed with the accuracy I am getting for my application.  So far everything has been bench tested and the STM32 and USFS powered from the USB port of my PC.  

I finally got the project into a plastic enclosure and added an 18650 battery.  The heading jumped by about 30 degrees and is no longer linear as the USFS is rotated 360 degrees.  This is strictly from the introduction of the battery. 

I have not used the unit on battery power long enough to see how the discharge of the battery might also influence the heading value.

Now for the questions:
1. Is calibration capable of taking into account the battery interference (hard iron I presume?)?

2. Do you have any best practices for using batteries in small portable applications with a magnetometer, or even recommended battery types?  

Thanks for your time,
Tom

kriswiner, 2020-12-10:
Hi Tom,

My guess is that the battery is ferromagnetic or at least is causing extra
magnetic field inside the enclosure. You might try a simple 1S 3.7 V LiPo
battery to see if the results are different and keep all metal away from
the magnetometer.

You can move the entire assembly around to try to recalibrate the
magnetometer; the EM7180 has dynamic mag cal so this might return the
device to accuracy...

gerowtj, 2020-12-10:
Thanks for the quick reply. 

I will try another style of battery and I'll experiment more with moving the unit around to see if it will recalibrate with different battery styles.  

I'll report back the results.

Thanks again!