kriswiner/EM7180_SENtral_sensor_hub #72: EM7180_set_WS_params() hangs at check parameter acknowledge register loop

K-Space909, 2021-03-05:
Hello! I had this running on the Teensy with no problem, but recently compiled for a SAMD Cortex M0+ board (Seeeduino Xiao), and the function EM7180_set_WS_params() now does not come out of this loop:

        while (!(STAT == param))
    {
        STAT = readByte(EM7180_ADDRESS, EM7180_ParamAcknowledge);
    }

Maybe I need to change something to work better with my board, but is this perhaps a communication issue between the Sentral and the EEPROM device?

kriswiner, 2021-03-05:
Could be, but if it works on a Teensy but not the SAMD then it sounds more
like a SAMD software problem to me. Do you get the expect output before
this hang?

K-Space909, 2021-03-05:
Yes I do. I get all of the normal ...Hold up- suddenly it works and I didn't change anything. Perhaps just a fluke.

K-Space909, 2021-03-05:
It got through the loop but it appears it did not load the WS parameters correctly. Hardware output is working but yaw is way off. Software outputs are unchanging at (declination), 0.00, 0.00

kriswiner, 2021-03-05:
Bad connections?

K-Space909, 2021-03-05:
Not sure, but I think my I2C connection is fine. Sorry to be a bother. Here's my debug outputs (I've been shuffling things around for my purposes)

        Initiating IMU
        Scanning...
        I2C device found at address 0x28  !
        done
        
        EM7180 ROM Version: 0xE69
        Should be: 0xE609
        EM7180 RAM Version: 0x17435
        EM7180 ProductID: 0x80 Should be: 0x80
        EM7180 RevisionID: 0x2 Should be: 0x02
        A barometer is installed
        A temperature sensor is installed
        EEPROM detected on the sensor bus!
        EEPROM uploaded config file!
        EEPROM upload successful!
        Mag Cal Found.
        Applying Mag Cal.
        Mag Cal applied.
        Accel Cal Found.
        X-acc max: 2052
        Y-acc max: 2046
        Z-acc max: 2131
        X-acc min: -2044
        Y-acc min: -2051
        Z-acc min: -1997
        Accel Cal applied.
        Beginning Parameter Adjustments
        Magnetometer Default Full Scale Range: +/-1000uT
        Accelerometer Default Full Scale Range: +/-8g
        Gyroscope Default Full Scale Range: +/-2000dps
        Magnetometer New Full Scale Range: +/-1000uT
        Accelerometer New Full Scale Range: +/-8g
        Gyroscope New Full Scale Range: +/-2000dps
         EM7180 mag calibration completed
         EM7180 magnetic anomaly detected
         EM7180 new quaternion result
         EM7180 new mag result
         EM7180 new accel result
         EM7180 new gyro result
         EM7180 sensor status = 0
        Actual MagRate = 100 Hz
        Actual AccelRate = 200 Hz
        Actual GyroRate = 200 Hz
        Actual BaroRate = 50 Hz
        
        *******************************************
        Send '1' to store Warm Start configuration
        *******************************************
        
        Ultimate Sensor Fusion Sentral/MPU9250 Init Complete!

kriswiner, 2021-03-05:
Looks fine to me...

K-Space909, 2021-03-05:
I'll try to just recalibrate. I did move it over a couple spots on the breadboard, and I noticed this thing is quite sensitive to surroundings. Very awesome work, Kris. I'm blown away after struggling with the BNO055. I would love to ask you a few random questions via email if possible. I have a project that I-m going to be taking into limited production, likely in the thousands. All open-source hardware and firmware.

closed 2021-03-05 by K-Space909

kriswiner, 2021-03-05:
tleracorp@gmail.com